LinuxCNC/linuxcnc #22: joints_axes: hotkeys no longer select joints/axes in Axis GUI

SebKuzminsky, 2016-01-06:
Ever since commit f2551bbaf05a8b41b0d162a964d82f57e94566af ("joints_axes9: joint jogging and axis gui updates") in joints_axes9, the keyboard can no longer select joints and axes in the Manual Control (F3) tab.

jepler, 2016-01-06:
This appears to be ref 6171b65a895d in branch ja10.

In my testing, I found that in sim/axis/axis.ini, XYZ select axes and backtick, 1, 2 (etc, henceforth "digits") set feed override.  This is a change compared to master branch, where digits select axes.

In sim/axis/ldelta.ini in joint mode, XYZ and digits all do nothing in joint mode.  Having homed and entered cartesian mode, XYZ and digits behave the same as for axis.ini, either selecting axes or feed setting override.

 1. is there a good reason to change the behavior of backtick, 1, 2 to select axes in cartesian mode?
 1. is it just an accident that `12 doesn't select joints in joint mode?

SebKuzminsky, 2016-01-06:
You must have used the "Home All" feature to home?  Hominga single joint by selecting it and hitting the Home key does not work (see #23)

jepler, 2016-01-06:
yup

SebKuzminsky, 2016-01-07:
Fixed by Dewey in 7f68abe (joints_axes10).

closed 2016-01-07 by SebKuzminsky