LinuxCNC/linuxcnc #23: joints _axes: cannot home in Axis GUI

SebKuzminsky, 2016-01-06:
Ever since commit f2551bbaf05a8b41b0d162a964d82f57e94566af ("joints_axes9: joint jogging and axis gui updates"), i can not home the joints of my non-trivkins robot arm.  I select the joint I want to home by clicking on it with the mouse, then hit the Home button on my keyboard, and Axis prints this:

```
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1437, in __call__
    return self.func(*args)
  File "/home/seb/linuxcnc-dev/bin/axis", line 2471, in home_joint
    if s.homed[trajcoordinates.index(vars.ja_rbutton.get())]:
ValueError: substring not found
```

The code in question is the home_joint function:

```
    def home_joint(event=None):
        if not manual_ok(): return
        doHoming=True
        if s.homed[trajcoordinates.index(vars.ja_rbutton.get())]:
            doHoming=prompt_areyousure(_("Warning"),_("This joint is already homed, are you sure you want to re-home?"))
        if doHoming:
            ensure_mode(linuxcnc.MODE_MANUAL)
            go_home(trajcoordinates.index(vars.ja_rbutton.get()))
```

The problem is that the vars.ja_rbutton.get() function returns "0" when I have joint 0 selected and am in Free mode, but it's trying to match is with an axis name (like "X" or "Y") from the ini file.

SebKuzminsky, 2016-01-06:
Jeff identified the matching commit in ja10 as 6171b65.

SebKuzminsky, 2016-01-06:
As noted in #22: Home All works, but homing an individual joint does not work due to this issue.

SebKuzminsky, 2016-01-07:
Fixed by Dewey in 7f68abe5431 (joints_axes10).

cradek, 2016-08-18:
Reopening.

ctjctj on irc reports this is happening on ubuntu16/amd64 in 2.7 when trying to home Y.  He reports that opening a gcode file fixes homing of Y.

I can't find either 6171b65 or 7f68abe.

Machinekit has a related fix, b188548, but it seems incomplete because this idiom is used in more than one place (homing and touch off?)

I think we assumed this was a tcl version problem, but M. Buesch says it's an amd64 problem.  ctjctj reports it on amd64.

ctjctj's backtrace:

```
cjohnson@whiteboard:~$ linuxcnc linuxcnc/configs/jgro2/jgro2.ini
LINUXCNC - 2.7.5
Machine configuration directory is '/home/cjohnson/linuxcnc/configs/jgro2'
Machine configuration file is 'jgro2.ini'
Starting LinuxCNC...
Found file(lib): /home/cjohnson/git/linuxcnc-dev/lib/hallib/core_sim.hal
Note: Using POSIX non-realtime
Found file(lib): /home/cjohnson/git/linuxcnc-dev/lib/hallib/sim_spindle_encoder.hal
Found file(lib): /home/cjohnson/git/linuxcnc-dev/lib/hallib/axis_manualtoolchange.hal
Found file(lib): /home/cjohnson/git/linuxcnc-dev/lib/hallib/simulated_home.hal
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1540, in __call__
    return self.func(*args)
  File "/home/cjohnson/git/linuxcnc-dev/bin/axis", line 2356, in home_axis
    if s.homed["xyzabcuvw".index(vars.current_axis.get())]:
ValueError: substring not found
Shutting down and cleaning up LinuxCNC...
```

jepler, 2016-08-19:
The 'substring not found' bug is different, see #146

closed 2016-08-19 by jepler