LinuxCNC/linuxcnc #27: sserial: everything goes south after a single error

jepler, 2016-01-10:
Reportedly, in 2.6 hm2 sserial would survive transmission errors; in 2.7, it does not recover from even a single transmission error.  I'll be looking into this once I am able to set up a test system with sserial.

floppydisk525, 2016-01-10:
I don't know if this helps you or not but here are two 'results' from my 5i25/7i77 setup.  One from 2.6.8 (I think) and one from 2.7.x.  

I did upgrade the firmware for the 5i25/7i77 as 2.7.0 said that the firmware should be updated.  I forget the number now, but I had a 2 year old firmware (at least).  

2.6.8 - I saw this error at least 2x and ignored it:  http://imgur.com/gallery/JKg0bLS/

2.7.x - I saw this error all the time and it wouldn't let me ignore it:  http://pastebin.com/uJYPrWgD

micges, 2016-01-10:
Masking non-fatal errors should help, this patch ignore extra character errors

    --- a/src/hal/drivers/mesa-hostmot2/sserial.c
    +++ b/src/hal/drivers/mesa-hostmot2/sserial.c
    @@ -1729,7 +1729,7 @@ int hm2_sserial_check_errors(hostmot2_t *hm2, hm2_sserial_instance_t *inst){
         rtapi_u32 buff;
         int i,r;
         int err_flag = 0;
    -    rtapi_u32 err_mask = 0xFF00E1FF;
    +    rtapi_u32 err_mask = 0xFF00E1F7;
         const char *err_list[32] = {"CRC error", "Invalid cookie", "Overrun",
             "Timeout", "Extra character", "Serial Break Error", "Remote Fault",
             "Too many errors",

jepler, 2016-02-24:
Fixed in 2.7.4.

closed 2016-02-24 by jepler