LinuxCNC/linuxcnc #28: Axis GUI MDI history problem on Debian Jessie

SebKuzminsky, 2016-01-18:
This problem happens on Debian Jessie with the Axis GUI, master and 2.7.

Hit up-arrow to select a command from the MDI history and it populates the MDI-command text-input field with that command, but then you can't use right-arrow or left-arrow to move around the command string to edit it.

If you click in the text-input field you can edit the command like normal.

On Lucid with 2.7 it works as expected: up-arrow/down-arrow select a line from history, then left-arrow/right-arrow let you edit that command.

On Wheezy with master it behaves the same correct way as on lucid/2.7.

So it looks like some kind of jessie/tcl/tk/focus/hocus/pocus.

rene-dev, 2017-10-25:
has been fixed in https://github.com/LinuxCNC/linuxcnc/commit/e52f2419b87f5c2134731122631c50045d9a2089

SebKuzminsky, 2017-12-04:
I cherry-picked this fix from master to 2.7 and removed the debugging "puts".  Thanks @jepler for the fix and @rene-dev for the reminder to close this issue!

closed 2017-12-04 by SebKuzminsky