LinuxCNC/linuxcnc #29: Clarify install docs re: sim/uspace

SebKuzminsky, 2016-01-20:
Our install docs are not sufficiently clear about how you can run sim/uspace debs on vanilla kernels for simulating your gcode programs, but that's a useful and valuable way to run linuxcnc.

jepler, 2016-08-07:
we also need to discuss how uspace can be used with rtai and xenomai realtime in master

rredford, 2018-12-01:
There is rather ugly workaround to get linuxcnc installed on your system. From https://forum.linuxcnc.org/9-installing-linuxcnc/32941-linuxcnc-on-debian-stretch?start=30#96331 :

Error is : Required version of libboost-python is not in stretch repository.

Temporarily point to jessie repo to install such library.

sudo nano /etc/apt/sources.list with nano to contain:
	deb http://ftp.us.debian.org/debian jessie main
sudo apt-get update
sudo apt-get install linuxcnc-uspace

Now remove the pointer to jessie in sources.list
sudo apt-get update

rredford, 2018-12-02:
Mine is just from repos. I suspect if I build it on system the such workaround isn't needed. We might want to make 2 repo versions, one for linuxcnc system, and one for other linux systems, for sim only.

andypugh, 2024-03-25:
It's been more than 8 years, but...

closed 2024-03-25 by andypugh