LinuxCNC/linuxcnc #30: G53 doesn't work right with G92 and coordinate system rotation

jepler, 2016-01-22:
Tested in 2.6.11 and master-ish with configs/sim/axis:
 1. Start AXIS.  Power on.  Home all.  Clear G54, G92 offsets via menu
 1. MDI: G54
 1. MDI: G0 G53 X0 Y0
 1. MDI: G92 X1 / G10 L2 P1 R45 / G0 G53 X0 Y0

Expected behavior: no motion at step 4, because the machine was already at G53 X0 Y0 after step 3
Actual behavior: motion to machine coordinates (.2929, -.7071)

jepler, 2016-01-22:
Instead of G0 G53 X0 Y0, you can also G91 G0 X0 Y0, which should cause motion since it's in incremental distance mode with all distances zero.  It takes you to the same machine location as the G0 G53 version.
[image]

jepler, 2021-10-10:
Thanks!  The best way to get a change accepted into linuxcnc is by preparing a pull request. Patches added to issues are likely to get lost.

andypugh, 2022-04-05:
Seems to be fixed by #1297

closed 2022-04-05 by andypugh