LinuxCNC/linuxcnc #42: Robot Arm needed

FrankOcean11, 2016-02-29:
im actually looking for the robot arm files which i see this program can be used for but i do not see that specific folder maybe a little pointer please? im looking to control a arm that i made from scratch and diy servos i built as well which is what controls the arm, my arm arent being control manually but autonomous.

jepler, 2016-02-29:
The issue tracker is not a location to request help configuring LinuxCNC.  The forum at http://forum.linuxcnc.org is a better location.

closed 2016-02-29 by jepler

FrankOcean11, 2016-02-29:
maybe you did not understood, i wasn't asking you to guide me through configurations. Im asking for the folders containing the source files for the robot arm