LinuxCNC/linuxcnc #44: ja: motion's joint locking hal pins missing

SebKuzminsky, 2016-03-09:
Skunkworks reports on IRC:  http://tom-itx.no-ip.biz:81/~tom-itx/irc/logs/%23linuxcnc-devel/2016-03-09.html

## Here are the steps I follow to reproduce the issue:

 1. Start with a pre-ja config with at least one [AXIS_n]LOCKING_INDEXER=1, auto-convert to a ja config with update_ini.
 2. Note that the LOCKING_INDEXER setting got applied to the correct [JOINT_n] section.
 3. Start linuxcnc with the converted config.
 4. Linuxcnc does not create the requested joint.n.unlock and joint.n.is-unlocked HAL pins.

## This is what I expected to happen:
The pins should be created.

## This is what happened instead:
The pins were not created.

## It worked properly before this:
Works in 2.7.4 (pre-ja), though in this version, Motion creates locking pins for all rotary axes, whether or not they asked for them in the ini.

In JA, Motion expects a new load-time argument of unlock_joints_mask, which is not created by update_ini.  Motion only creates locking pins for the joints enabled in that mask.

Possible fixes would be to revert to the old system of creating locking pins for all rotary joints, or to ensure that unlock_joints_mask gets sent to Motion correctly at load time.

jepler, 2016-03-15:
Possible fixes also include showing the user a message that the inifile was detected as having features that could not be automatically converted.

jepler, 2016-03-15:
...or even just documenting that this is not automatically converted.

SebKuzminsky, 2016-04-26:
Dewey Garrett (the primary Joints/Axes developer) tells me in private email:

> commit 1c83eadd notes that
"2) less common items may not be converted
    (locking indexers, non-identity kinematics, wheel jogging, haltcl files)"

> I don't think updating for locking indexers should be a requirement -- there
cannot be many users of that functionality

andypugh, 2016-05-15:
The pins and INI entries are now created, commit 5e60946b798466ee1b03ff6b8423fc7e8dd94acd

closed 2016-05-15 by andypugh