LinuxCNC/linuxcnc #45: ja: Command isn't tracking feedback in machine-off

samcoinc, 2016-03-10:
## Here are the steps I follow to reproduce the issue:

 1.On a servo machine with encoder feedback - after machine is up / homed 
 2.Toggle machine back off 
 3.Physically move axis
 5.Toggle machine back on
 6.You instantly get a following air for that axis and machine powers back off.
 7.Currently no way to fix this state

On sim
 1.Power up and home
 2.Open halcmd
 3.unlink joint.0.motor-pos.fb
 4.setp joint.0.motor-pos-fb 5
 5.Toggle machine back on.
 6.At this point - fb should equal commanded..  It doesn't and you get a following error.

## This is what I expected to happen:
The machine should turn on and the fb should equal cmd position
## This is what happened instead:
Following error
## It worked properly before this:

If you do the exact same sequence in 2.7 - when you toggle power on - the machine turns on as expected.

samcoinc, 2016-03-10:
Ok - so the above directions seem to not work 100% of the time.  What does seem to work 100% of the time is 
1. power up - home machine
2. open halcmd
3. unlink joint.0.motor-pos-fb
4. jog X axis -> get a following error
5. toggle machine on. - > following error -> machine turns off.

jepler, 2016-03-11:
I think another connected issue is how ja doesn't update cartesian feedback position from joint feedback position while in "machine off" mode.  

1. power on, home all, **machine off**
2. open halcmd
3. unlink joint.0.motor-pos-fb
4. setp joint.0.motor-pos-fb various values

Steps 3/4 are emulating the case where you have a servo machine where you can turn the motor shafts with the amplifiers off, or if you have a nice big machine where the Z axis droops measurably with amplifiers off.

Expected: cartesian feedback position tracks changes in joint feedback, as it does in 2.7

I think that if this cartesian-tracks-joints behavior were happening, then the fix at 3cd7b92 would not be needed (but is probably harmless)

jepler, 2016-03-15:
I tested at the tip of ja12 and the steps I described in my previous comment now work as I expect.  @samcoinc also says things are better for him.

closed 2016-03-15 by jepler