LinuxCNC/linuxcnc #49: 2.6: abort during startup interrupts startup

SebKuzminsky, 2016-03-26:
Micges reports on IRC:

http://tom-itx.no-ip.biz:81/~tom-itx/irc/logs/%23linuxcnc-devel/2016-03-26.html#13:49:34

SebKuzminsky, 2016-06-06:
Jeff and I discussed a possible fix here:
http://tom-itx.no-ip.biz:81/~tom-itx/irc/logs/%23linuxcnc-devel/2016-06-06.html#09:42:01

zultron, 2016-06-13:
Seb, I've verified the fix for 2.6.  Nice work!

I picked the commits into another tree I'm working on, and everything works except `status.gcodes` isn't updated.  This actually causes the `statbuffer-g5x-abort` test to fail in that tree.  Running *any* MDI command (even `(foo)`) causes `status.gcodes` to be updated, though, so it looks like it should be something easy.

If you hold off on merging this a few days, I'll try and fix that problem, and will let you know if it's somehow relevant to LCNC, even though it looks like it isn't at this point.

SebKuzminsky, 2016-06-14:
Ok, i'll hold off until I hear from you.  Thanks for testing it.

zultron, 2016-06-16:
@SebKuzminsky, this is looking great.  The issue I had was related to the state tags work, so it doesn't apply to LinuxCNC.  I really appreciate the fix!

SebKuzminsky, 2016-06-16:
Thanks for looking into it and letting me know, i just pushed this to our 2.6 branch.

zultron, 2017-12-07:
The fix for this particular issue (abort during task init) was in 402c27b, but was reverted in 9f07c33.  The corresponding regression test `motion-logger/startup-gcode-abort` was disabled in bdffc97.

I'm almost done with some updates to the state tags branch that make a lot of improvements, and while I was at it, cherry-picked 402c27b back onto it, and the test seems to pass again.

I wonder if this one shouldn't be reopened until it's actually fixed (an org member will have to do it).  Otherwise I'll start a new issue pointing here.

andypugh, 2017-12-07:
Issue re-opened as requested

zultron, 2020-05-19:
State tags has been merged into `master` at this point.  I believe this issue is no longer outstanding, or to the extent that it is, it is superseded by #858, tracking intermittent failures in `motion-logger/startup-gcode-abort` test.

PR #67 (2016-06-16): Seb/2.6/task abort fix
This fixes #49 and the Task abort/statbuffer issue that [Zultron reported](http://thread.gmane.org/gmane.linux.distributions.emc.devel/16220).

I wish it wasn't so invasive, so late in the 2.6 and 2.7 stable branches, but it fixes real bugs with as little impact as I can manage.

This branch changes behavior in a couple of ways:
* Abort now cleans up Interp and Canon just about as much as a real M2 or M30 would.
* Task startup results in a bunch more Canon & Motion commands being processed.  These are the same commands it's always generated (just about), but now it doesn [... cut]
commit 402c27bd (2016-06-16): Task: add drain_interp_list

drain the interp_list after Task init and Abort.

Don't clear the interp_list on Abort, if we're draining.

Fixes #49.