LinuxCNC/linuxcnc #56: ja: auto-switch to world mode fails for some configs

pkmcnc, 2016-04-26:
## Here are the steps I follow to reproduce the issue:
Start the machine and home

## This is what I expected to happen:
Switch to world mode like hexapod-sim config does.

## This is what happened instead:
After homing the machine remains in joint mode. Otherwise everything seems to work.

## Information about my hardware and software:
Six axis genhexkins machine with analog servos.
 * I am using joints_axes13 from git on a standard Debian ISO
 * I am using AXIS
 * I am using MESA 7i77 & 5i25

SebKuzminsky, 2016-04-26:
Dewey Garrett (the primary Joints/Axes developer) tells me in private email:

> Cannot reproduce without more information
  1) need to know the commit SHA
  2) need inifile or better tar of entire config
  3) my guess: he did not use [KINS]AUTO_TELEOP=timelimit
     as described in the getting-started/updating-linuxcnc document
     (The swith is not automatic)

pkmcnc, 2016-04-26:
Thanks a lot!
I compiled without docs...
I'm sure AUTO_TELEOP will help.

closed 2016-04-26 by pkmcnc