LinuxCNC/linuxcnc #57: stepconf in 2.7.4 "axis test" popup forgets Velocity and Acceleration settings

SebKuzminsky, 2016-04-27:
## Here are the steps I follow to reproduce the issue:

1. Start StepConf, go through the configuration to the "X Axis Test" popup window.
1. Observe that Velocity is initialized to 1.0 in/s and Acceleration to 30 in/s^2. 
1. Change the Velocity and Acceleration settings, then click the Cancel button.
1. Click the "Axis X/Test This Axis" button again.
1. Observe that Velocity and Acceleration have reset to 1 in/s and 30 in/s^2, forgetting the settings you set.

## This is what I expected to happen:
The Velocity and Acceleration fields should remember what the user set them to.

## Information about my hardware and software:
Debian Wheezy from the linuxcnc install iso, Linux 3.4-9-rtai-686-pae, linuxcnc-2.7.4 from www.linuxcnc.org.

SebKuzminsky, 2016-04-27:
Same behavior if you click "Ok" in step 3.

nicokid, 2016-06-22:
Hi,
This problem is only for the X-axis or also for all other axis (y,z, etc.)?

jepler, 2016-09-11:
Hm.

Here's what I would have thought should happen when closing the axis test:
 * Clicking OK: Copy acceleration and velocity values back into the axis page
 * Clicking Cancel: Don't change any values on the axis page
.. and in any case, when starting the axis test, the accel/vel values shown on the axis page are used.

This way, OK means "these new values are great, I want to use them" and cancel means "These new values are trash, I want to keep the old values".

Thoughts?

SebKuzminsky, 2016-09-11:
I think @jepler is right: Cancel should discard any changes made, and Ok should preserve them by copying them from the "X Axis Test" popup back to the "Axis X" page of the Stepconf main window.  My bug report was in error for claiming Cancel should preserve them.

Current behavior (as of 2.7.7, probably in 2.6 and master too but I haven't tested those versions) is that clicking Ok in the "X Axis Test" popup erroneously does not copy the values back to the matching fields in the "Axis X" page.  Clicking Cancel returns to the main window without copying the values, as it should.

SebKuzminsky, 2016-09-11:
@nicokid I verified that this affects X, Y, and Z axes (in 2.7.7).