LinuxCNC/linuxcnc #59: make it easier to change uspace rtapi_app

SebKuzminsky, 2016-05-05:
Currently (as of 2.7.4) the only way to change rtapi's logging message level is to change the source code and recompile.  It would be better if it was

SebKuzminsky, 2016-05-05:
RTAI's realtime logging is controlled through /proc, which is also awkward.

It would be good if we could control all rtapi implementations' msg_level in some consistent way, perhaps by a new NML message to Motion, or new Motion HAL pins.

jepler, 2016-05-05:
.. and that /proc code is disabled for kernels 3.10 and greater too

jmkasunich, 2016-08-07:
Regarding Seb's suggestion that rtapi's message level be controlled by an NML message to motion, or by motion HAL pins - please consider that some people use HAL without motion.  A halcmd command to change the debug level would be more appropriate.  Also, it is theoretically possible to use RTAPI for things other than HAL.  (Although I doubt anyone does.)  So even using halcmd to set the level wouldn't satisfy a purist...  I'm not that much of a purist, but I don't think that a method involving the motion component is appropriate.

andypugh, 2016-08-09:
Can you elaborate on the disablement of /proc ? Is that a kernel thing or a LinuxCNC thing?
I have got used to echoing a number to somewhere in /proc to change the reporting level.

jepler, 2016-08-09:
In the distant past, RTAPI + RTAI created entries in /proc.  By kernel 3.10, the APIs for doing so changed incompatibly, and nobody updated the code to continue working -- I just disabled it instead.   (ec38ac4566e07baa3e47e0a0463ac7384c92f3aa).  But creating /proc entries is simply not possible for uspace-style realtime, so it's not a good solution for that either.

andypugh, 2024-03-25:
I think that I fixed this, without closing the issue, in 2020 via 07dcd21

closed 2024-03-25 by andypugh