LinuxCNC/linuxcnc #60: homing bug in 2.7.4

SebKuzminsky, 2016-05-05:
JT reports on IRC:

http://tom-itx.no-ip.biz:81/~tom-itx/irc/logs/%23linuxcnc-devel/2016-05-05.html#15:06:29

< JT-JA13> testing Dewey's homing branch based on JA and if I set my HOME_SEARCH_VEL = 0.1 and 
                    my HOME_LATCH_VEL = 0.05 I get Home switch active before start of latch move
< JT-JA13> I remember Andy saying something once about needing more search velocity for certain 
                    homing 
< JT-JA13> So I tried HOME_SEARCH_VEL = 0.5 and HOME_SEARCH_VEL = 0.5 and it worked
< JT-JA13> I also tested in 2.7.4
< JT-JA13> just to make sure it was in both

jepler, 2016-08-07:
There doesn't seem to be enough information to reproduce this.  Is it possible to demonstrate the problem with a (modified) sim config?

petterreinholdtsen, 2023-10-03:
As no new information has shown up for nine years, I suggest to close this issue.

closed 2023-10-04 by andypugh