LinuxCNC/linuxcnc #64: tkemc button help-version fails

isladelobos, 2016-06-03:
The button help---->about in tkemc show a error: (cant read "env(EMC2VERSION)": no such variable

SebKuzminsky, 2016-06-03:
Thanks for the bug report.

I just pushed a commit that I believe fixes this.  Please try version v2.6.12-25 or newer from the buildbot (http://buildbot.linuxcnc.org/) and let us know if it fixes the issue for you.

May I ask, why are you using tklinuxcnc?  I'm not arguing, just trying to understand.  Is it because it has some feature or behavior that is not available in the other GUIs, or because it's what you're used to?  Or some other reason?

The original authors of tklinuxcnc are no longer active in the LinuxCNC project, so maintenance of this program has falled on us other developers.  Some of us are GUI developers, though none of us focus on tklinuxcnc as our main passion.  Have you looked at some of the newer, more well-maintained GUIs like Axis, Gmoccapy, and Gscreen?

SebKuzminsky, 2016-06-04:
2.6.12-25 fixes the issue for me (I can reproduce it on 2.6.12 but not on 2.6.12-25).

commit f1f7b278 (2016-06-03): tklinuxcnc UI: rebranding

Get the LinuxCNC version number from the rebranded environment variable
(LINUXCNCVERSION), not the old one (EMC2VERSION).

Fixes #64.
closed 2016-06-08 by SebKuzminsky