LinuxCNC/linuxcnc #76: Consolidate "how to build" docs

SebKuzminsky, 2016-06-17:
We currently have instructions on how to build LinuxCNC in at least four different places:
* /README.md
* /docs/INSTALL
* http://wiki.linuxcnc.org/cgi-bin/wiki.pl?Installing_LinuxCNC
* http://wiki.linuxcnc.org/cgi-bin/wiki.pl?Installing_EMC2

Plus many distro-specific build instructions linked from the Installing_LinuxCNC page...

I believe it should be easy to distill these instructions down to a small recipe that works on all our supported platforms, because the buildbot does just that.  The instructions should be in our asciidocs, so that they are tracked in git and automatically published on our website.  The build instructions should be in the "Developer Information" section, right after "Contributing to LinuxCNC".

The top-level README should not duplicate the information, but should probably point to both the published docs online and to the asciidoc source document in the git working dir.

jepler, 2016-07-15:
@SebKuzminsky  is your work on this in master branch enough to make you feel this issue can be closed?

SebKuzminsky, 2016-07-15:
No, i only did the fun easy part.  I still have to do the tedious bits:

[x] create the new Building LinuxCNC document
[] Update /README.md
[] Update /docs/INSTALL
[] Update all the various wiki pages to point to the html docs on www.linuxcnc.org

luzpaz, 2023-06-26:
Is this ticket still relevant ?

SebKuzminsky, 2023-06-26:
Thanks for the reminder about this ticket.  Since I haven't gotten to those two unchecked checkboxes of remaining work in the past 7 years it seems unlikely that I'll get to them anytime soon... So I'll close this ticket and we can open new tickets for any old build instructions we come across.

closed 2023-06-26 by SebKuzminsky