LinuxCNC/linuxcnc #80: MDI of an arc that passes through the soft limit - overshoots soft limits by de-acceleration amount.

samcoinc, 2016-06-19:
# The issue tracker is not a support forum

## Here are the steps I follow to reproduce the issue:

Sim Axis.

 1. Move Y axis 5 inches away from positive soft limit.  Zero
 2. X axis should be somewhere in the middle of travel an zeroed
 3. G0X10Y0
 4. g2x10i10f300

## This is what I expected to happen:

A linear movement checks to see of the end point is outside of the soft limits.  Some similar check should happen for arcs.  

## This is what happened instead:

The machine moves towards the soft limit - hits the soft limit *then* de-accelerates.  On our Matsuura - the over travel is far enough to hit the over travel limit switches in this situation.  (the above example- the axsis moves past the soft limit by .393"

## It worked properly before this:

This is 2.7.4 - I don't think this has ever worked.

jepler, 2016-06-20:
As pointed out by @cradek this was also reported on the old bug tracker: https://sourceforge.net/p/emc/bugs/130/

We currently only check the endpoint of an arc for being inside the work volume https://github.com/LinuxCNC/linuxcnc/blob/19aaa6f752/src/emc/task/emctaskmain.cc#L502

As long as the work volume is a simple axis-aligned rectilinear solid, it might suffice to find which of 0, 90, 180, and 270 degrees (the cardinal directions) are within the arc, and then test that all those points are in the soft limits.

SebKuzminsky, 2016-06-20:
... Except for `G10 L2 R`.

SebKuzminsky, 2016-07-15:
Some hand-wavy discussion here: http://tom-itx.no-ip.biz:81/~tom-itx/irc/logs/%23linuxcnc-devel/2016-07-15.html#11:39:49