LinuxCNC/linuxcnc #82: Gmoccapy creates gcode files with mode 0600 instead of obeying umask

lair82, 2016-06-23:
## Here are the steps I follow to reproduce the issue:

 1. Create gcode program on remote computer
 2. Drop gcode program into nc_files folder on CNC over network
 3. Load gcode program in Gmoccapy to run the program, 
4. Open program in editor, and make any edit to program, then save.
5. Go back to remote computer and try to access/open gcode program, Access Denied

## This is what I expected to happen:
I should be able to open for viewing or copy the program back out of the CNC to save to remote computer for future use, 

## This is what happened instead:
Access is denied because the editor is for some reason changing the permissions to not allow anyone other than the CNC control to access that program.

## It worked properly before this:
It works perfectly on my 3 Ubuntu machines, but on my 2 Wheezy machines, it does not work at all.

## Information about my hardware and software:

 * Using Debian Wheezy, 
 * Kernel version: Linux greenmill 4.1.13-rt15 #1 SMP PREEMPT RT Fri Dec 4 14:31:34 EST 2015 i686 GNU/Linux
 * I am running ...
   * [X] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * LinuxCNC version: 2.7.3-2.7.4
 * User interface: Gmoccapy

jepler, 2016-06-23:
@gmoccapy can you please take a look at this?  I believe this occurs because of hal_sourceview.py:safe_write, which uses tempfile.mkstemp, which is documented as creating a file that only the current user can read.

gmoccapy, 2016-06-25:
Hallo Jeff.

You are right, it is caused by hal_sourceview.py

I found a solution, just adding after line 260 os.chmod (filename, 
mode). Before pushing, we should agree about a default value for mode.

Do you think 0664 is OK as default?

Norbert (gmoccapy)

jepler, 2016-06-25:
@gmoccapy if you will not be able to change it so that the user's umask is respected (by using a different API than tempfile.mkstemp to create the temporary file), then 0664 is probably not the worst choice.

jepler, 2016-06-25:
In other words, that's fine.

gmoccapy, 2016-07-11:
I fixed that with commit f102b9f118b52346c17a4c3f966327dcf0fc5127

GladeVCP hal_sourceview, so it was not a gmoccapy bug!!
Please close issue!
Norbert

jepler, 2016-07-11:
We'll have to manually close this issue once the next 2.7 release is made.

jepler, 2016-07-14:
2.7.5 has been released.

closed 2016-07-14 by jepler