LinuxCNC/linuxcnc #86: entry fields blank on stepconf axis pages

jepler, 2016-06-27:
## Here are the steps I follow to reproduce the issue:

 1. open stepconf
 2. go forward to the X Axis page

## This is what I expected to happen:
The numbers in the entry fields should be visible

## This is what happened instead:
The numbers are invisible

## It worked properly before this:
Merging pull request #71.  The problem also exists at the tip of #33. I have worked around the problem by disabling `axis_sanity_test` in commit 44c5fb919 on our master branch.

## Information about my hardware and software:

 * Debian 8 amd64
 * Kernel  4.4.6-1~bpo8+1 
 * I am running a binary I built myself, v2.8.0-pre1-1771-g928b7f2
 * XFCE desktop environment with default ("xfce") theme

nicokid, 2016-08-31:
This workaround is unnecessary . The problem existed only when there were not the correct .glade files.
Nicola.

jepler, 2016-08-31:
@nicokid thanks for noting that!  If you prepare a pull request that reverts ffc5fb9 and double check that the axis pages work properly, I'll merge it.  I can't do that testing right this second.

nicokid, 2016-09-01:
Ok @jepler I'm controlling also the issue #57.
When I solved the problem #57 I do a pull of both.

closed 2024-10-29 by jepler