LinuxCNC/linuxcnc #97: joint vs axis soft limits

SebKuzminsky, 2016-06-29:
In master after the JA merge.

On a trivkins machine (a gantry), a misconfiguration resulted in an axis and its joint having different soft limits.  The axis soft limit was outside the joint soft limit.  I could jog into the joint soft limit, resulting in an error message:

```
    Exceeded POSITIVE soft limit (0.05000) on joint 2
    Hint: switch to joint mode to jog off soft limit
```

On other axes/joints the soft limits coincided (as they should) and the machine would smoothly decelerate to a halt at the limit, rather than run over it and error.

Probably *both* axis and joint soft limits should be honored, and whichever is more restrictive should control the machine's work envelope.

rene-dev, 2016-07-28:
There is a lot more to this problem, in world mode all the joint limits(acc/vel) are ignored.
The problem is the planner only plans in cartesian space, and is not aware of the kinematic.
Really the planner should put the result trough the kinematic, check for joint limits, and then put it back trough the kinematic to check if there is an ambiguity.
Currently this fails with a joint following error, because the joint will just overspeed.
But yes, a work envelope constrained by joint limits would be super nice.

jepler, 2016-07-28:
@rene-dev Yes, obeying joint constraints (velocity and acceleration) while running in Cartesian mode is a major area for future work.