LinuxCNC/linuxcnc #99: After Update to 2.7.4 error warning when using "Run From Line" in Gmoccapy

lair82, 2016-06-30:
## Here are the steps I follow to reproduce the issue:

 1. Press Run from line
 2. Enter line number you would like to start from, press enter
 3. Error pops up
 4. Select OK
 5. Press "Cycle Start"
 6. Machine starts at selected line

## This is what I expected to happen: Machine should start running at selected line

## This is what happened instead: An error pops up, the machine still runs at the selected line, it is just giving this fault when trying to perform the command, here is a screenshot of the fault

https://app.box.com/s/8yoc1u3b6auxkeixzmczz2xbcm2fgnn6

## It worked properly before this: After upgrading from 2.6.4 to 2.7.4

## Information about my hardware and software:

 * Linux Version: Ubuntu 10.04
 * I am running ...
   * [x] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * LinuxCNC version : 2.7.4
 * Interface: Gmoccapy

gmoccapy, 2016-07-11:
Has been fixed already in April 2016 (11.04.2016),
so can be closed!

ver. 1.5.6.2
- bugfix: run from line was trying to set an entry in the alarm page, which
 I droped in 1.5.6 
 I cleared the code from unneeded imports and comments

jepler, 2016-07-11:
@gmoccapy just to confirm, the fix is in 2.7 branch? If so we'll close this issue when the next 2.7 release is made.

gmoccapy, 2016-07-11:
Hallo Jeff,

Yes it is in the 2.6, 2.7 and master, it has been fixed with commit     
f102b9f118b52346c17a4c3f966327dcf0fc5127

Norbert

jepler, 2016-07-14:
2.7.5 has been released.

closed 2016-07-14 by jepler