LinuxCNC/linuxcnc #100: Error compiling taskintf

TurBoss, 2016-07-01:
Hi

Configure output: http://dpaste.com/2HAGKHS
Error: http://dpaste.com/0EB8BRS

I'm having this error when compiling master branch 3e93bfa14ecf1653b8c38ef6d1c1f51c330148ca  on ubuntu 16.04

i'm building a simulator with --with-realtime=uspace

Thx

SebKuzminsky, 2016-07-01:
Looks like a conflict in the definition of isnan() between the standard C++ header file "cmath" and boost.

Try the attached patch, which tells it to prefer the cmath implementation over the boost one.

[isnan-patch.txt](https://github.com/LinuxCNC/linuxcnc/files/342738/isnan-patch.txt)

closed 2016-07-01 by cradek

jepler, 2016-07-01:
Looks like we regressed, this was fixed once before at commit 9d6a5bc0.  If the  patch isn't adequate, please re-open this issue with the next compile error if it looks like it's also about `isnan`, or open a new issue if it looks like something else.

TurBoss, 2016-07-01:
it has passed the but still compiling fine

will tell if finish

TurBoss, 2016-07-01:
now I'm getting this

http://dpaste.com/0GED3Z0

TurBoss, 2016-07-01:
sry can't reopen so I'm goin to create a new one

commit f4eb3750 (2016-07-01): taskintf: fix compile error reported on on Ubuntu 16.04

We fixed this back at commit 9d6a5bc but with the merge of
joints-axes it appers some cases of it were added back in.

I can't test, but it looks good by eyeball.

Closes #100.