LinuxCNC/linuxcnc #103: small bugs in Axis

pkmcnc, 2016-07-03:
## Here are the steps I follow to reproduce the issue:
 1.Start hexapod-sim config
 2.Home
 3.Press Control-Space OR press $

## This is what I expected to happen:
Clean pop up messages OR switch to joint mode

## This is what happened instead:
Messages are cleaned but also axis selection disappears.
OR
$ doesn't switch the mode from the first attempt

## It worked properly before this:
Works correct in 2.7.4

## Information about my hardware and software:
 * I am using Debian from ISO in VirtualBox
 * I am running a binary I built myself
 * I am using master LinuxCNC from git, just updated
 * I am using AXIS

SebKuzminsky, 2016-07-03:
In the current tip of the 2.7 I did this:

1. Start sim/axis/vismach/hexapod-sim/hexapod.ini
1. Hit F1, then F2
1. Click the "Home All" button
1. Observe that the machine homes all joints and all joints get the "homed" target icon in the DRO
1. Hit $ key to switch from Joint to World mode
1. Observe that the "joints" DRO is replaced with the "axes" DRO.

When doing the same sequence on the master branch the behavior is different: after "Home All" finishes, the machine automatically switches to World mode, indicated by the "joints" DRO being replaced by the "axes" DRO.

In this state the "$" key still switches back and forth between Joint and World mode.  It's just that the new default for a homed machine is to be in World mode.

This was an intentional change, we felt that it was an improvement over the old behavior because most users would want to be in World mode whenever possible, now that incremental jogging and wheel jogging works in World mode.

Did we miss some reason to be in Joint mode after homing?

pkmcnc, 2016-07-03:
I'm well aware of auto teleop, it's a great feature!
But $ should be pressed twice to switch to joint mode after homing. Then it works as supposed. Probably it's not aware that auto teleop already switched the mode? So it switches to teleop (but we are already there) and the next press switches to joint.

SebKuzminsky, 2016-07-03:
Ah, I see what you're saying, I misunderstood your bug report before.  Yes, i reproduce that behavior in master (c2aaae25): after Home All and auto-teleop, hitting $ once does nothing, hitting $ again switches to Joint mode.  That does indeed seem like a bug, I'll look into it.

SebKuzminsky, 2016-07-03:
Dewey fixed this in  f9a817ef86 (thanks!), but i just noticed a related bug.  After Home All and auto-teleop, the "View" menu's options of "Joint mode" and "World mode" still shows the marker by "Joint mode", which is not right.  I think there's a mismatch between s.motion_mode and the vars.teleop_mode that sort of tries to mirror it.

pkmcnc, 2016-07-03:
I pulled the latest commits and the bug is gone for most non-identity configs: hexapod-sim, max5kins, lineardeltakins
But I also have 5axes/5joints sim config with custom kinematics, and it still has the bug. I'll dig more to reproduce it.

pkmcnc, 2016-07-25:
Fixed

closed 2016-07-25 by pkmcnc