LinuxCNC/linuxcnc #104: Ctrl-Space clears axis selection in AXIS for non-identity configs after homing

pkmcnc, 2016-07-03:
## Here are the steps I follow to reproduce the issue:
 1.Start hexapod-sim config
 2.Home (the machine goes to world mode, axis X radiobutton is selested by default in manual control tab).
 3.Press Ctrl-Space to clean pop up messages

## This is what I expected to happen:
Clean pop up messages, if any.

## This is what happened instead:
Messages disapper but axis selection is also lost, no axis radiobutton is active.

## It worked properly before this:
Works correct in 2.7.4 

## Information about my hardware and software:
 * I am using Debian from ISO in VirtualBox
 * I am running a binary I built myself
 * I am using master LinuxCNC from git, just updated
 * I am using AXIS

jepler, 2016-07-03:
What appears to be happening is, a non-visible radiobutton widget for some joint has keyboard focus.  control-space activates that radiobutton, and also clears messages.  Normally this effect goes unnoticed.

You can see this more clearly if you do this sequence of actions:
 1. press X to select the X axis
 1. press tab.  A black border moves from the "X" radiobutton to the "Y" radiobutton
 1. press ctrl-space.  The active radiobutton moves from "X" to "Y", and messages are cleared.

The reason this appears to be a regression since 2.7.4 is that the selected joint radiobutton is hidden, but ctrl-space still is passed to that widget.

pkmcnc, 2016-07-03:
https://github.com/LinuxCNC/linuxcnc/commit/b6dac597d74dcfb6b17ce78c3d08a417e9adadaa fixed that, thanks!

closed 2016-07-03 by pkmcnc