LinuxCNC/linuxcnc #111: Teach Interp to reject arcs that reach outside the axis soft limits

SebKuzminsky, 2016-07-15:
The interpreter currently (as of master 37e6fc45c) only verifies that the endpoints of arcs are within the axis soft limits.  The curve of the arc may extend outside the soft limits, as long as it returns to within the soft limits before ending.

You can verify this behavior by running sim/axis/axis (which has a -X limit of -10), F1, F2, Home All, G0 X-9, F100, G2 Y6 J3.

The arc is accepted, and travels outside the axis soft limits (though also see issue #96).

The extents of the arc are easy to determine and validate against the axis soft limits.  Coordinate system rotation may require extra care.

jepler, 2016-07-15:
I think this is a duplicate of #80.  Close if you agree.

SebKuzminsky, 2016-07-15:
Derp, dupe.

closed 2016-07-15 by SebKuzminsky