LinuxCNC/linuxcnc #114: TLO affects live plot in Axis

pkmcnc, 2016-07-19:
## Here are the steps I follow to reproduce the issue:

 1. Start a machine (axis_mm). Home.
 2. Edit TLO in the tooltable for a tool #1 to -200mm, save and reload the table.
 3. M6 T1 G43.
 4. G53 G0 Z0.
 5. Touch off Z to 0. Now both relative and machine Z = 0.

## This is what I expected to happen:
The program preview should remain at Z=0 inside the limits box.

## This is what happened instead:
The preview is far above the box.

## It worked properly before this:
Probably never, in 2.7.5 and master it's as described.

pkmcnc, 2016-10-21:
Thanks for the fix! I'll test asap. 
Is it going to master too?

jepler, 2016-10-21:
@pkmcnc yes it will be merged up to master.  We "merge up" from release branches into master, just not always right away.  It looks like in this case there's a conflict merging 2.7 to master due to some other changes by @SebKuzminsky -- when he gets those squared away, this will get merged too.

pkmcnc, 2016-10-22:
Just tested it in master, works as expected. Thank you!
I also observed some false "Program exceeds..." behavior from time to time, so I'm glad that this series of commits is supposed to fix it too. Thanks again!

closed 2016-10-22 by pkmcnc