LinuxCNC/linuxcnc #115: image2gcode: incompatible with numpy >= 1.9

jepler, 2016-07-19:
image2gcode relies on numpy.numarray, which has been removed from current distributions.  This will affect us in Debian Stretch, and already affects bleeding-edge users on Ubuntu 16.04 and Mint 18: http://docs.scipy.org/doc/numpy/reference/routines.numarray.html

blu-base, 2019-11-06:
Is this issue still open for some reason?

I was looking into the the respective file (src/emc/usr_intf/axis/scripts/image-to-gcode.py) to find that this issue has been fixed long time ago see 3d2aec7b18a7b286e15a8f34649ce3dbf9cbf3a5

Is there something obvious left to do which i missed?

zultron, 2020-06-10:
As @blu-base suggests, close this one out!  One less item on the 2.8 milestone:

https://github.com/LinuxCNC/linuxcnc/milestone/1