LinuxCNC/linuxcnc #116: loading a program changes optional stop setting

gmoccapy, 2016-07-22:
Hallo,

found another strange behavior:

  * start sim/axis_mm.ini
  * start hal show and watch halui.program.optional-stop.is-on (it is on
    by default here)
  * push the optional stops button (pin should be False by now)
  * load a program of your choice (i.e. arcspiral.ngc)
  * See the halui pin, it has changed

As far as I tested, that happen every time a new program is loaded (mode 
changes to mode 2).

IMHO that is not correct, it should be possible to set the 
optional_stops in any motion mode at any time and the setting should 
remain set, until a user changes that.

Norbert

jepler, 2016-07-22:
I used gdb to find where optional stop is being set back to true.  It happens while task is processing a message of type EMC_TASK_PLAN_INIT

~~~~
Hardware watchpoint 3: *$1

Old value = false
New value = true
(gdb) where
#0  INIT_CANON () at emc/task/emccanon.cc:2474
#1  0x00007f712c3ee65a in Interp::init (this=0x11b8c50)
    at emc/rs274ngc/rs274ngc_pre.cc:814
#2  0x0000000000413bca in emcTaskPlanInit () at emc/task/emctask.cc:440
#3  0x000000000041e77f in emcTaskIssueCommand (cmd=0x118d7f0)
    at emc/task/emctaskmain.cc:2301
#4  0x0000000000411993 in emcTaskPlan () at emc/task/emctaskmain.cc:1228
#5  main (argc=<optimized out>, argv=<optimized out>)
    at emc/task/emctaskmain.cc:3309
~~~~

All UIs using the Python linuxcnc module recently had their behavior change, by adding a call to reset_interpreter to program_open.  At the time I made this change, I didn't realize this consequence of it (see commit 27d3822f):
~~~~
static PyObject *reset_interpreter(pyCommandChannel *s, PyObject *o) {
    EMC_TASK_PLAN_INIT m;
    emcSendCommand(s, m);
    Py_INCREF(Py_None);
    return Py_None;
}

static PyObject *program_open(pyCommandChannel *s, PyObject *o) {
    if(!reset_interpreter(s, o))
        return NULL;

    EMC_TASK_PLAN_OPEN m;
    char *file;
    int len;

    if(!PyArg_ParseTuple(o, "s#", &file, &len)) return NULL;
    if(unsigned(len) > sizeof(m.file) - 1) {
        PyErr_Format(PyExc_ValueError, "File name limited to %zu characters", sizeof(m.file) - 1);
        return NULL;
    }
    strcpy(m.file, file);
    emcSendCommand(s, m);
    Py_INCREF(Py_None);
    return Py_None;
}
~~~~

closed 2016-07-23 by cradek

pkmcnc, 2016-07-25:
https://github.com/LinuxCNC/linuxcnc/commit/92ca9a20cc2e19facfb31d7066340acc1120f3ae also fixed a dangerous bug with spontaneous TLO reset.

commit 92ca9a20 (2016-07-23): linuxcncmodule: don't reset interpreter on program_open

this had side effects like resetting the optional stop flag.

Closes #116