LinuxCNC/linuxcnc #117: UI becomes unresponsive after hitting limit switch

jepler, 2016-07-22:
## Here are the steps I follow to reproduce the issue:

 1. Use the attached config (it requires no particular hardware): [limit_test-sim.zip](https://github.com/LinuxCNC/linuxcnc/files/379149/limit_test-sim.zip)
 2. start linuxcnc, home all axes, jog (I used right arrow) and while jogging, click the "trip limit" button
 3. press F1 or otherwise try to take the machine back out of estop

## This is what I expected to happen:
linuxcnc works

## This is what happened instead:
the UI is unreponsive until exit

## It worked properly before this:
This broke between 2.7.4 and 2.7.5.  My bisect indicates that it broke at 402c27bdd, which means it may also affect 2.6 branch.

## Information about my hardware and software:

 * debian jessie amd64 with vanilla kernel (also reported on other systems)
 * A binary I built myself
 * v2.7.5
 * AXIS UI
 * simulated parport

jepler, 2016-07-22:
In emcSendCommand echo_serial_number is never advancing.  it's stuck (in this case) at 23
~~~~
(gdb) p ((EMC_STAT*)s->s->get_address())->echo_serial_number
$7 = 23
(gdb) p s->serial
$8 = 30
~~~~
task state (emcStatus->status) is rcs_exec (2), so task thinks it's still doing .. something.

2.7.4 prints an interesting message before recovering:
~~~~
Cannot unhome while moving, joint 0
~~~~
is it possible that task is *not* fully clearing out an ongoing jog, and this ongoing jog is what prevents the jog from "ending"?

jepler, 2016-07-22:
@SebKuzminsky please let me know if you don't reproduce it with this config file.  @jethornton thanks for the config, it was easy enough to adapt to sim_parport.

c-morley, 2016-07-23:
I just had a similar problem on my lathe. Here I found the MPG was counting by it's self do to a grounding issue.
This caused the axis to jog immediately at machine on, soon the ui was lagging behind commends.
It ran till estop or limit switch was tripped.
I assume the jogging flooded NML commands?
strangely after  I would get that  error 'can't unhome while axis 2 is moving'
Maybe this helps?
Chris M

SebKuzminsky, 2016-07-27:
I believe this is fixed in v2.6.12-63 and v2.7.5-31.  I'd appreciate feedback from @jethornton and @c-morley, let me know if the issue is resolved for you in those versions.

commit ef5dc9d7 (2016-07-27): motion: when motion disables, mark all joints as "in position"

Task sometimes waits for Motion to be "done" rather than "execing".

This commit fixes a bug in Motion where it would sometimes forget to
tell Task that it's done (aka in position, aka not moving any more).

This fixes #117.
closed 2016-07-28 by cradek