LinuxCNC/linuxcnc #122: hal_glib - All homed not working as expected on not trivial kinematics

gmoccapy, 2016-07-24:
lib/python/hal_glib.py do emit signals if linuxcnc.stat attributes does change.

But homing is done with axis mask, but homing is done in Joint mode.

If you have a XYZAC config with JOINT 0 to 4, you will get gremlin homed all 5 Joints, but not AXIS C, so the "all homed" signal will not be emitted.

If you have a identity machine a work around is just changing [KINS] JOINTS to 6 and [TRAJ] AXES = 6
and it will work, because the joint jog button will no be shown in gmoccapy. But if you do the same work around on non IDENTITY machines, you will get 6 Joints and that is not correct.

To follow my mind disaster, please get the branch 

gmoccapy_JA_based_on_master

try sim/gmoccapy/gmoccapy_XYZAC.ini it does work as expected with 6 AXES and 6 JOINTS, having only AXES and JOINTS

so now try sim/gmoccapy/gmoccapy-JA/table-rotary-tilting/xyzac-trt.ini

You will not get out of JOINT mode, as hal_glib will never emit the all homed signal.
pressing the home all button will emit self.command.home(-1) and linuxcnc will home all 5 joints, but hal_glib expect C, (being axis 6) to be homed to emit the corresponding signal.

I tried to change hal_glib.py to check for joints homed, but documentation of the python interface have not been updated jet, so I not even know if something like stat.joints_mask does exist.

Can you give me a hint where to begin?

Norbert

gmoccapy, 2016-07-24:
I think I found a solution, please give me some time to test all the other GUI.
If that works, I will push the changes.

Norbert

gmoccapy, 2016-07-24:
Pushed solution with commit 091c64537f3d9a747cce75d30908825a0e5b84a1

So please close the issue.

Norbert

jepler, 2016-07-24:
Thanks @gmoccapy

closed 2016-07-24 by jepler