LinuxCNC/linuxcnc #130: MDI Error reset the feed value

gmoccapy, 2016-08-01:
## Here are the steps I follow to reproduce the issue:

 1. out of estop, home machine and change to MDI
 2. enter G0X20
 3. enter F300
 4. enter G2 X0Y0 I-5        (I know that this causes an error! see end and start does not match, OK)
 5. enter again G2 X0Y0 I-5 (Now you get can't make a circle with feed 0

## This is what I expected to happen:
Same error, as nothing has changed

## This is what happened instead:
The error has reseted the Feed Value F300, I do not see any reason for that
and if that happens, the active gcode should be updated, but it stays with F300 untill you execute the next MDI command 

## It worked properly before this:
it is fine in LinuxCNC 2.7.6

happens with axis and gmoccapy!

SebKuzminsky, 2016-08-30:
I'm assuming this bug report is against master, since you say it works in 2.7.6.

I think it is related to my faulty Task/Abort changes, somewhat documented in #126.

gmoccapy, 2016-09-01:
Sorry for the late answer,

Yes it is against master.

Norbert

gmoccapy, 2016-09-25:
Still the same behavior!