LinuxCNC/linuxcnc #139: hal pin not updated correcttly on start up

gmoccapy, 2016-08-14:
I have an external estop and a momentary switch to switch on my machine.
This is the relevant part of the hal file:
`###############################################
#  Estop circuit and machine enable signals   #
#            connect them to pins             #
#           (c) Norbert Schechner             #
###############################################

# if the whatchdog has bit, we must enable estop
# that is why we include watchdog in our estop circuit
loadrt not          names=watchdog
addf   watchdog     servo-thread
net    dog_sig_in   watchdog.in                   <=   hm2_7i76e.0.watchdog.has_bit
net    dog_sig_out  watchdog.out                  =>   hm2_7i76e.0.7i76.0.0.output-08

# ESTOP Switch is between output-08 and input-20 to signal LinuxCNC estop is OK
# you may have to connect additional hardware to ensure security, like termal sensor switches, etc

# if OK we set an output to signal we are ready to enable the machine
# use output to enable the external allowing to enable the machine
net    estop_ext    iocontrol.0.emc-enable-in     <=   hm2_7i76e.0.7i76.0.0.input-20
net    estop_out    iocontrol.0.user-enable-out   =>   hm2_7i76e.0.7i76.0.0.output-09

# Enable button is connected between output-09 and input-21

# machine enable signals
# enable the servo amplifieres or stepper drivers with this output
net    ena-machine  hm2_7i76e.0.7i76.0.0.input-21 =>   halui.machine.on
net    enabled      motion.motion-enabled         =>   hm2_7i76e.0.7i76.0.0.output-10
`

Whatchdog.has_bit is connected to not.in
not.out to output-08
output-08 to estop-switch
estop-switch to input-20
input-20 to iocontrol.0.emc-enable-in
iocontrol.0.emc-enable-out to output-09
output-09 to momentary-switch
momentary-switch to input-21
input-21 to halui.machine.on

If I start the machine with pressed estop (estop circuit is open / broken)
and get out of estop, closing the circuit, output-09 get TRUE and I can use the momentary switch to enable my machine.

If I start my machine with ESTOP circuit being OK, then output-09 will not get hight, output-08 is hight. I need to push estop, release it again to update the hal pin state.

IMHO that behavior is not OK.

Norbert

jepler, 2016-08-14:
@gmoccapy what component are you saying is buggy?  What inputs or input-transitions does that component treat in a way you think is improper?

btw I fixed your markup.  Use a line with just "~~~~" before and after code blocks, and remember you can "preview" before submitting and I believe you can also edit the text after submitting

gmoccapy, 2016-08-20:
Hallo Jepler,

IMHO something with the hm2_eth driver is not OK, as it does not update the hal pin correctly, it only updates the outputs on a change, but that does not happen on start up.

Norbert

pcw-mesa, 2016-08-20:
You really cannot use the FPGA I/O watchdog this way. If you have a watchdog event, its basically 
fatal and you must restart LinuxCNC (you certainly cannot turn on output 8 since after a watchdog bite
All I/O will be disabled and in a OFF state

gmoccapy, 2016-08-20:
@pcw-mesa,

OK I understand, but if I change the config to 

setp hm2_7i76e.0.7i76.0.0.output-08 1

I have the same behavior, it should set input 20 to True if i start with estop circuit closed and through iocontrol.0.user... it should set output 09 to TRUE, but that does not happen!

Norbert

pcw-mesa, 2016-08-20:
I can't duplicate this 
any initial state of I/O gets copied properly to outputs
I have a test setup with 7I76E outputs looped back to 7I76E inputs with wires
 (0 to 0, 1 to 1 etc) either starting condition works as expected:

setp  hm2_7i76e.0.7i76.0.0.output-00 false
net io0 hm2_7i76e.0.7i76.0.0.input-00-not hm2_7i76e.0.7i76.0.0.output-01
net io1 hm2_7i76e.0.7i76.0.0.input-01-not hm2_7i76e.0.7i76.0.0.output-02
net io2 hm2_7i76e.0.7i76.0.0.input-02-not hm2_7i76e.0.7i76.0.0.output-03
net io3 hm2_7i76e.0.7i76.0.0.input-03-not hm2_7i76e.0.7i76.0.0.output-04
 works
as does
setp  hm2_7i76e.0.7i76.0.0.output-00 true
net io0 hm2_7i76e.0.7i76.0.0.input-00-not hm2_7i76e.0.7i76.0.0.output-01
net io1 hm2_7i76e.0.7i76.0.0.input-01-not hm2_7i76e.0.7i76.0.0.output-02
net io2 hm2_7i76e.0.7i76.0.0.input-02-not hm2_7i76e.0.7i76.0.0.output-03
net io3 hm2_7i76e.0.7i76.0.0.input-03-not hm2_7i76e.0.7i76.0.0.output-04

Do you check that output 9 is actually set in your failing case?

gmoccapy, 2016-08-21:
I will do some more testing and come back to that issue.

Norbert

gmoccapy, 2016-08-24:
Hallo,

I followed the signals in detail and found the problem to be iocontrol.

connected as described in first post!

If I start the machine with ESTOP enabled (open circuit)
the signal runs fine until iocontrol.0.emc-enable-in

So all OK here, now I close the ESTOP circuit (reset estop) and the signal runs through iocontrol to output-09, so that is fine!

If I start linuxcnc with ESTOP in restet mode (circuit closed) I do expect the signal to reach also output-09, but it only reaches iocontrol.0.emc-enable-in.

iocontrol.0.emc-enable-out does not get high!

opening and closing the ESTOP circuit solved the problem, but IMHO if in is hight, out should be hight too.

Bug or race conflict?

Norbert

rene-dev, 2023-09-26:
probably closed by https://github.com/LinuxCNC/linuxcnc/pull/2497

closed 2023-09-26 by rene-dev