LinuxCNC/linuxcnc #157: Wrong DRO after start up

gmoccapy, 2016-08-21:
## Here are the steps I follow to reproduce the issue:

 1. change sim/axis/axis_mm.ini to have home positions at 
 2.
 3.

## This is what I expected to happen:

## This is what happened instead:

## It worked properly before this:
(If the behavior changed after making a particular change in hardware or
software, describe the change you think is responsible.  E.g., "after upgrading
from LinuxCNC 2.7.3 to 2.7.4")

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
 * I am using this kernel version (shown by `uname -a`):
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):

cradek, 2016-08-22:
Here is what I did to test:

Checked out and built v2.6.12-76-g588b186

I tested changing [AXIS_1]HOME=2.2222 and with G54 Y offset=0, after homing the DRO shows 2.2222 as expected (and shows we are in G54 with G54 Y=0).  I added a G54 Y offset with MDI G10 L2 P1 Y1.  Now after restarting AXIS and homing, the DRO shows G54 Y=1 as expected.  When in Relative mode Y shows 1.2222, when in Machine mode Y shows 2.2222, both as expected.  This is all correct as far as I can tell AFTER commit 85210a3.  It WAS broken as you are describing (or similar) BEFORE commit 85210a3.  Are you sure about this bug report, in particular which revisions are bad and good?  Are my steps to reproduce it wrong?

If I have not reproduced it, please tell exactly what variables to change in the sim/axis/axis_mm.ini file, and exactly what revision you are running.

gmoccapy, 2016-08-22:
I just tested again with the most recent 2.7 branch, it is still wrong.
I will make a config for you with a step to step instruction to follow that.

Just give me some time!

Meanwhile you can follow my step by step in the first post and you will see that it is not OK!

Norbert

cradek, 2016-08-22:
I just tried the same tests I described above in v2.7.6-14-gdf183a4 and it also works how I expect, same as described above.  I will wait for more information, thanks.

gmoccapy, 2016-08-24:
Hallo Chris,

I made some more testing all with the same result, I followed your way and also got wrong DRO, so I went on analysing and after I made a git reset -- hard origin/2.7 a make clean and compiled completely new, the problem was solved.

So the issue is solved. Sorry for reporting "my mixed up repo" instead of a< real bug.

Please close this issue.

Sorry Norbert

cradek, 2016-08-24:
Thanks, Norbert.  It's so easy to have mixups, no problem.

closed 2016-08-24 by cradek