LinuxCNC/linuxcnc #162: `/tests/interp/compile` test creates junk file

zultron, 2016-09-14:
Running the `interp/compile` test creates a junk (apparently `.var`) file:

```
$ runtests tests/interp/compile
Running test: tests/interp/compile
Runtest: 1 tests run, 1 successful, 0 failed + 0 expected
$ /bin/ls -a tests/interp/compile | grep '^[^Retu.]' | od --format=x1
0000000 c8 e3 53 da 7f 0a
0000006
$ head -3 $'tests/interp/compile/\xc8\xe3\x53\xda\x7f'
5161	0.000000
5162	0.000000
5163	0.000000
```

This is true for at least the 2.7 branch.  Especially annoying when using both `runtests` and `git` in a development workflow.

SebKuzminsky, 2016-09-14:
I'm not able to reproduce this problem.  I'm testing with 88e22fa (which is just after 2.7.7) on Debian Stretch.

```
$ git clean -fdx tests/
$ runtests tests/interp/compile/
Running test: tests/interp/compile
Runtest: 1 tests run, 1 successful, 0 failed + 0 expected
$ /bin/ls -1a tests/interp/compile/
.
..
expected
.gitignore
README
rs274ngc.var
test.sh
use-rs274
use-rs274.cc
$ /bin/ls -1a tests/interp/compile/ | /bin/grep '^[^Retu.]'
rs274ngc.var
```

Is there any chance that the file with the strange name was created by some other means?  If you remove it by hand, is it re-created when you run the test again?

jepler, 2016-09-14:
I am seeing something similar to @zultron .  Debian Jessie, 2.7 branch at
~~~~
70a9a36 Docs: fix example scrips so they work when copied and pasted.
~~~~

jepler, 2016-09-14:
~~~~
$ strace -ff -e open ./use-rs274  
....
open("h;\330j\177", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 4
~~~~

jepler, 2016-09-14:
valgrind:
~~~~
Syscall param open(filename) points to uninitialised byte(s)
  at 0x5ABAC00: __open_nocancel (syscall-template.S:81)
  by 0x5A53CA1: _IO_file_open (fileops.c:227)
  by 0x5A53E19: _IO_file_fopen@@GLIBC_2.2.5 (fileops.c:332)
  by 0x5A49063: __fopen_internal (iofopen.c:90)
  by 0x4F69708: Interp::save_parameters(char const*, double const*) (rs274ngc_pre.cc:1860)
  by 0x4F6A33B: Interp::synch() (rs274ngc_pre.cc:1962)
  by 0x4F6CDB9: Interp::init() (rs274ngc_pre.cc:1163)
  by 0x4029DD: main (in /home/jepler/src/linuxcnc/tests/interp/compile/use-rs274)
Address 0xffeffdc81 is on thread 1's stack
in frame #5, created by Interp::synch() (rs274ngc_pre.cc:1928)
~~~~

zultron, 2016-09-14:
I haven't tested, but sweet! Looks like @jepler nailed it.  Thanks fellas.

commit 0316a08f (2016-09-14): testsuite: specify var filename

closes #162.
closed 2016-09-15 by SebKuzminsky