LinuxCNC/linuxcnc #166: Spindle Increase/Decrease

jethornton, 2016-09-24:
Spindle increase/decrease in src/emc/motion/command.c has a FIXME for spindle increase/decrease. This would be a benefit to some users. https://forum.linuxcnc.org/forum/24-hal-components/31609-increases-decreases-spindle-speed

gmoccapy, 2016-09-25:
For override there is a pin halui.spindle-override.scale adding such a pin should solve that FIXME.

Norbert

andypugh, 2016-09-25:
Norbert: That's a different thing. This request is for a change in the step-size when the + / - buttons are used in GUIs which use the "spindle increase" and "spindle decrease" commands. 

I am not sure why those commands exist, to be honest. It probably makes more sense for a GUI to keep its own number and send that as an absolute command. I am pretty sure that G-code doesn't use them.

gmoccapy, 2016-09-25:
IMHO it is very similar.
Just insert a pin halui.spindle-inrement and read that value in src/emc/motion/command.c as increment.
So the user can set what ever he wants. If no value is set, we should stay with the 100 U/min

But you are right, the best would be just delete that part and make command.c stay clean, As you sayed it is a matter of the GUI to take care of that values.

By the way, gmoccapy 2.nn has new sliders and takes care of that ;-)

Norbert

c-morley, 2016-09-25:
The problem is that command.c doesn't seem to have status for spindle increase increment.
or more rightly 	emcmotStatus and    emcmotCommand  doesn't have it - though I haven't searched right though all the message events.

hmm looking in emc_nml.hh it seems EMC_SPINDLE_INCREASE can get a speed setting.
I just haven't figured out how the NML message gets turned into a motion message...

fix that part then any GUI could set it's increment as part of the message to increase.
Then the GUI could decide how to get that setting anyway it wants.

I fixed a similar thing in AXIS a couple years back so that spindle on could set the default speed from an INI setting. In that case the problem was in a higher level (emcmodule.cc)

Chris M

cradek, 2016-09-25:
I agree with a lot of what cmorley says.  I don't agree with the FIXME in the code.  It makes no sense to have "increase now" come from the GUI and "how much" come from HAL.

If we need increase/decrease NML messages, the amount should come in the message from the GUI.  But there is no reason to have these messages at all.

These increase/decrease messages are from the pre-HAL days, probably invented for mshaver's bridgeport with an airmotor on the varispeed, and no speed feedback other than your eyeballs looking at the dial on the head.  Those bridgeports originally had go-faster and go-slower buttons, and you'd hold them down to make the airmotor crank the varispeed up or down while watching the dial, and release the button at the right time.  This was duplicated in keystick, and later xemc, and those buttons probably just set output bits (on the parport or STG) while they were held down, like the original control.

Today we'd use HAL to do that, and there would not be an NML message for increase/decrease at all.  Today the single message type "turn the spindle [forward/backward] at [command] speed" is what makes sense.  If the GUI wants increase/decrease buttons, it can just add or subtract from the current speed and send a new message.