LinuxCNC/linuxcnc #178: No way to correctly display back tool lathe in axis

propcoder, 2016-10-12:
## Here are the steps I follow to reproduce the issue:

 1. Take sample config Sim Axis Lathe.
 2. Try to change the config to reverse X axis in display (extents, toolpath, liveplot, tool shape preview) and in keyboard key mapping. What I tried is parameters in INI [DISPLAY] section:
`LATHE = 1
GEOMETRY = -XZ
JOG_AXES = ZX
JOG_INVERT = X`
and / or .axisrc changes as described in, probably for versions 2.7 and earlier:
[http://wiki.linuxcnc.org/cgi-bin/wiki.pl?BackToolLathe].(url)

## This is what I expected to happen:
Correctly reversed X axis as mentioned above.

## This is what happened instead:
In LinuxCNC 2.7.7 wiki method helps except it does not display tool shape at all.
In LinuxCNC/AXIS version 2.8.0-pre1-2648-gd1919c5 only part of objects mentioned are reversed correctly.

rene-dev, 2017-07-13:
I got it working, by forcing axis into y2 view, and disable the x invert.
the remaining issue is the tool preview, which does not work in y2 mode.
also, the grid does not seem to work in y2.
but I could not find where the preview is generated, can anyone point me in the right direction?
I also propose a change to the ini file:
use BACK_TOOL_LATHE = 1 like gmoccapy for back tool lathes.
I also send this to emc-devel, but it did not seem to have worked...

Rene

closed 2017-07-13 by cradek

MZot, 2020-06-06:
I am not sure that this is the right place to post this, but I have a fix for tool shape preview in Y2 view. Can someone please test.
in glcanon.py:

```
    def lathetool(self, current_tool):
        glDepthFunc(GL_ALWAYS)
        diameter, frontangle, backangle, orientation = current_tool[-4:]
        w = 3/8.

        radius = self.to_internal_linear_unit(diameter) / 2.
        glColor3f(*self.colors['lathetool'])
        glBegin(GL_LINES)
        glVertex3f(-radius/2.0,0.0,0.0)
        glVertex3f(radius/2.0,0.0,0.0)
        glVertex3f(0.0,0.0,-radius/2.0)
        glVertex3f(0.0,0.0,radius/2.0)
        glEnd()

        glNormal3f(0,1,0)

        if orientation == 9:
            glBegin(GL_TRIANGLE_FAN)
            for i in range(37):
                t = i * math.pi / 18
                glVertex3f(radius * math.cos(t), 0.0, radius * math.sin(t))
            glEnd()
        else:
            dx, dy = self.lathe_shapes[orientation]

            min_angle = min(backangle, frontangle) * math.pi / 180
            max_angle = max(backangle, frontangle) * math.pi / 180

            sinmax = math.sin(max_angle)
            cosmax = math.cos(max_angle)
            tanmax = math.cos(max_angle)
            sinmin = math.sin(min_angle)
            cosmin = math.cos(min_angle)
            tanmin = math.cos(min_angle)

            circleminangle = - math.pi/2 + min_angle
            circlemaxangle = - 3*math.pi/2 + max_angle
            d0 = 0

            x1 = (w - d0)

            sz = max(w, 3*radius)

            glBegin(GL_TRIANGLE_FAN)
            glVertex3f(
                radius * dx + radius * math.sin(circleminangle) + sz * sinmin,
                0,
                radius * dy + radius * math.cos(circleminangle) + sz * cosmin)
            for i in range(37):
                #t = circleminangle + i * (circlemaxangle - circleminangle)/36.
                t = circleminangle + i * (circlemaxangle - circleminangle)/36.
                glVertex3f(radius*dx + radius * math.sin(t), 0.0, radius*dy + radius * math.cos(t))

            glVertex3f(
                radius * dx + radius * math.sin(circlemaxangle) + sz * sinmax,
                0,
                radius * dy + radius * math.cos(circlemaxangle) + sz * cosmax)

            glEnd()
            glBegin(GL_TRIANGLE_FAN)  #y2 triangle, BACK_TOOL_LATHE
            glVertex3f(
                radius * dx + radius * math.sin(circlemaxangle) + sz * sinmax,
                0,
                radius * dy + radius * math.cos(circlemaxangle) + sz * cosmax)
            for i in range(37):
                t = circlemaxangle - i * (circlemaxangle - circleminangle)/36.
                glVertex3f(radius*dx + radius * math.sin(t), 0.0, radius*dy + radius * math.cos(t))
            glVertex3f(
                radius * dx + radius * math.sin(circleminangle) + sz * sinmin,
                0,
                radius * dy + radius * math.cos(circleminangle) + sz * cosmin)
            

            glEnd()
        glDepthFunc(GL_LESS)
```

rene-dev, 2020-06-07:
This has been fixed a long time ago, can you explain what the problem is?

MZot, 2020-06-07:
version 2.7.15, gmoccapy, gremlin preview. Tool is not visible with BACK_TOOL_LATHE = 1. 
It really looks like it is fixed in master. Why was it never pushed to 2.7?

commit 4290ecf8 (2017-07-13): Merge branch 'master' of https://github.com/rene-dev/linuxcnc

Closes: #178