LinuxCNC/linuxcnc #179: Incorrect Axis view mode

phillc54, 2016-10-13:
## Here are the steps I follow to reproduce the issue:

 1. In an existing 4 axis configuration with LATHE = 0 in the [DISPLAY] section

## This is what I expected to happen:
Axis to display in perspective view
jog_order to be XYZA

## This is what happened instead:
Axis displays in a rotated Y view
jog_order is ZXYA
Y axis is not shown in the Manual Control tab

## It worked properly before this:
Upgrading to 2.8.0-pre1-2648-gd1919c5 from previous 2.8.0.... from buildbot.linuxcnc.org

SebKuzminsky, 2016-10-20:
I'd like to know what the previous version of 2.8.0 you had was, where things worked right.  You can probably see the old version number by running this command:

    grep 'upgrade linuxcnc' /var/log/dpkg.log

I'd also like to see your whole config, can you attach your ini file to a comment here?

jepler, 2016-10-24:
It will probably also be necessary to look at your specific config file.  The config files from before the conversion to JA will also be useful to diagnose if the problem turns out to be in the config converter.

I modified the sample config `by_interface/parport/stepper-xyza/inch.ini` to be sim-only by changing standard_pinout.hal as follows:
~~~~
-loadrt hal_parport cfg="0x0378"
+loadrt sim_parport names=parport.0
~~~~
and things look like I expect them to.

phillc54, 2016-10-25:
Oops, I may have led you up the garden path.

I have not used Axis for a while and only did recently after this update 
so I may NOT have had the 'LATHE =' line in my config previously.

I have just tried on 2.7.7 and it exhibits the same behavior so there 
has probably been no change in operation.

Maybe the docs should be changed to reflect the fact that "LATHE = 
"ANYTHING" will change the view to lathe.

Sorry about the false alarm.

Cheers, Phill

closed 2017-04-28 by phillc54