LinuxCNC/linuxcnc #180: python interface

fixerslo, 2016-10-16:
1. When in MDI mode, `linuxcnc.stat.settings[1]` returns wrong value if in G95 mode after spindle is started:

-Issue G95 in MDI
-set feed with F word
-turn on spindle with M3 S

## This is what I expected to happen:
1. `linuxcnc.stat.settings[1]` should return the value that was set with F word

## This is what happened instead:
1.  `linuxcnc.stat.feed_hold_enabled` returns value, that is 60 times greater than value that was set with F.  When in AUTO mode it works correctly

## It worked properly before this:

## Information about my hardware and software:

 LinuxCNC 2.7.4 
Gmoccapy GUI
metric mode

SebKuzminsky, 2016-10-20:
I can replicate this problem with LinuxCNC 2.7.7.

I started sim/axis/axis, F1 F2 Ctrl-Home, switch to the MDI tab, G21, G95,
F10 (10 mm/rev).

Then in another window I ran python, imported the linuxcnc module, instantiated linuxcnc.stat, polled, and got settings = (0.0, 10.0, 0.0).  Back in Axis I switched to the Manual tab, jogged around a little, switched back to MDI and ran 'S1000 M3'.  Back in python i polled linuxcnc.stat again and got settings = (0.0, 600.0, 1000.0).  In Axis MDI tab, the 'Active G-Codes' also shows F600.

SebKuzminsky, 2016-10-20:
Another data point: same version of linuxcnc, same (imperial) config, F1 F2 Ctrl-Home, **no** G21, G95, F10, jog around, then S100 makes the Active G-Codes F-word and linuxcnc.stat.settings[1] jump to an unexpected value.

SebKuzminsky, 2016-10-20:
There are two bugs here.  One is that the F-word changes surprisingly, and the other is that you have to issue an MDI command (after Aborting, or switching into and back out of Manual) before the wonky F-word is revealed.

commit c4b63616 (2016-10-20): canon: return correct feed rate in G95 mode

Canon keeps track of the current feed rate in the variable
currentLinearFeedRate.  This gets set by the canon call SET_FEED_RATE(),
which handles G95 ("Units per Revolution") mode differently from G94
("Units per Minute").

Interp::synch() re-synchronizes Interp from Canon.  In order for synch()
to work right when G95 is set, GET_EXTERNAL_FEED_RATE() needs to also know
to treat currentLinearFeedRate differently depending on the feed mode.

This closes #180.  Thanks to @fixerslo on github for this bug report.
closed 2016-10-22 by cradek