LinuxCNC/linuxcnc #181: Building LinuxCNC

CNC-432, 2016-10-16:
Hi! I'm trying to build LinuxCNC. It's failing at the linking stage.

## Here are the steps I follow to reproduce the issue:

> ./autogen.sh
>
> CFLAGS="-O2 -fPIC" \
> CXXFLAGS="-O2 -fPIC" \
> ./configure \
>   --prefix=/usr \
>   --libdir=/usr/lib64 \
>   --sysconfdir=/etc \
>   --localstatedir=/var \
>   --mandir=/usr/man \
>   --docdir=/usr/doc/linuxcnc-2.7.7 \
>   --enable-shared \
>   --disable-static \
>   --enable-non-distributable=yes \
>   --with-realtime=uspace \
>   --with-x \
>   --build=x86_64-slackware-linux
>
> make

[log-configure-linuxcnc.txt](https://github.com/LinuxCNC/linuxcnc/files/531766/log-configure-linuxcnc.txt)
[log-make-linuxcnc.txt](https://github.com/LinuxCNC/linuxcnc/files/531767/log-make-linuxcnc.txt)

## Information about my hardware and software:

 * I am using this Linux distribution and version: Slackware 14.1
 * I am using this kernel version: Linux 3.10.17 x86_64
 * I am using this LinuxCNC version: linuxcnc-2.7.7.tar.gz

jepler, 2016-10-16:
The error is:
~~~~
Linking rs274
/usr/lib64/gcc/x86_64-slackware-linux/4.8.2/../../../../lib64/libreadline.so: undefined reference to `tgetnum'
~~~~
similar to https://bugzilla.redhat.com/show_bug.cgi?id=499837 -- As noted there, Debian and debian-derived) platforms do not have this problem due to different decisions about library linking.

The fix would involve setting an appropriate value for `READLINE_LIBS` in `configure`.
I will be happy to review any proposed pull request, but I don't personally use slackware or have access to any slackware systems.

CNC-432, 2016-10-18:
Thanks for clarifying what is causing this.

## Patch:

> --- a/linuxcnc-2.7.7/src/emc/sai/Submakefile	2016-09-08 04:12:38.000000000 +0300
> +++ b/linuxcnc-2.7.7/src/emc/sai/Submakefile	2016-10-18 17:31:38.000000000 +0300
> @@ -3,7 +3,7 @@
>  SAISRCS := $(addprefix emc/sai/, saicanon.cc driver.cc dummyemcstat.cc) \
>  	emc/rs274ngc/tool_parse.cc emc/task/taskmodule.cc emc/task/taskclass.cc
>  USERSRCS += $(SAISRCS)
> -LIBREADLINE=-lreadline
> +LIBREADLINE=$(READLINE_LIBS)
>  
>  ../bin/rs274: $(call TOOBJS, $(SAISRCS)) ../lib/librs274.so.0 ../lib/liblinuxcnc.a ../lib/libnml.so.0 \
>  	../lib/liblinuxcnchal.so.0 ../lib/liblinuxcncini.so.0 ../lib/libpyplugin.so.0

closed 2016-10-18 by CNC-432

jepler, 2016-10-18:
Thanks.  Fixed in our master branch.

commit b6dc604b (2016-10-18): rs274: use configure-detected readline libs

The hardcoded value of -lreadline reportedly fails on
Slackware.  Close #181.