LinuxCNC/linuxcnc #182: AXIS/Gmoccapy acting like axis is homed when it is un-homed (2.8~pre)

gorbypark, 2016-10-17:
## Here are the steps I follow to reproduce the issue:

 1. Open StepConf
 2. Basically click forward all the way through a new config.  I selected to use Gmoccapy and changed pins for axis A to unused, inverted axis Y and Z and set max velocities on the 3 axis'.  All other options as per default.
 3. Run LinuxCNC 2.8~pre (compiled October 16, 2016) and try to move axis manually to home position.  Axis' will not move past position 0.0, even when completely un-homed (red DRO in gmoccapy) after starting LinuxCNC.  This prevents the machine from moving to it's home position.
 4. Manually switched the .ini to use Axis with the same results

## This is what I expected to happen:

Be able to move the position of any axis to any position while the axis is un-homed.

## This is what happened instead:

Axis' stop at 0.0 (basically where ever the machine happens to be when LinuxCNC starts).  Moving to a positive position works, but moving backwards it stops at 0.0

## It worked properly before this:

Works in 2.7.7

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Debian GNU/Linux 7.11 Wheezy
 * I am using this kernel version (shown by `uname -a`): Linux linuxcnc 3.4-9-rtai-686-pae #1 SMP PREEMPT Debian 3.4.55-4linuxcnc i686 GNU/Linux
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [x] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): v2.8.0-pre1-2653-ge8e4116
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): Gmoccapy and AXIS
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): Parallel port

> # Generated by stepconf 1.1 at Sun Oct 16 20:24:46 2016
> # If you make changes to this file, they will be
> # overwritten when you run stepconf again
> 
> [EMC]
> MACHINE = X2-2.8
> DEBUG = 0
> VERSION = 1.0
> 
> [DISPLAY]
> DISPLAY = axis
> EDITOR = gedit
> POSITION_OFFSET = RELATIVE
> POSITION_FEEDBACK = ACTUAL
> ARCDIVISION = 64
> GRIDS = 10mm 20mm 50mm 100mm 1in 2in 5in 10in
> MAX_FEED_OVERRIDE = 1.2
> MIN_SPINDLE_OVERRIDE = 0.5
> MAX_SPINDLE_OVERRIDE = 1.2
> DEFAULT_LINEAR_VELOCITY = 0.10
> MIN_LINEAR_VELOCITY = 0
> MAX_LINEAR_VELOCITY = 0.35
> INTRO_GRAPHIC = linuxcnc.gif
> INTRO_TIME = 5
> PROGRAM_PREFIX = /home/mike/linuxcnc/nc_files
> INCREMENTS = .1in .05in .01in .005in .001in .0005in .0001in
> 
> [KINS]
> JOINTS = 3
> KINEMATICS = trivkins coordinates=XYZ
> 
> [FILTER]
> PROGRAM_EXTENSION = .png,.gif,.jpg Greyscale Depth Image
> PROGRAM_EXTENSION = .py Python Script
> png = image-to-gcode
> gif = image-to-gcode
> jpg = image-to-gcode
> py = python
> 
> [TASK]
> TASK = milltask
> CYCLE_TIME = 0.010
> 
> [RS274NGC]
> PARAMETER_FILE = linuxcnc.var
> 
> [EMCMOT]
> EMCMOT = motmod
> COMM_TIMEOUT = 1.0
> COMM_WAIT = 0.010
> BASE_PERIOD = 100000
> SERVO_PERIOD = 1000000
> 
> [HAL]
> HALFILE = X2-2.8.hal
> HALFILE = custom.hal
> POSTGUI_HALFILE = postgui_call_list.hal
> 
> [TRAJ]
> COORDINATES =  X Y Z
> LINEAR_UNITS = inch
> ANGULAR_UNITS = degree
> CYCLE_TIME = 0.010
> DEFAULT_LINEAR_VELOCITY = 0.10
> MAX_LINEAR_VELOCITY = 0.35
> 
> [EMCIO]
> EMCIO = io
> CYCLE_TIME = 0.100
> TOOL_TABLE = tool.tbl
> 
> [AXIS_X]
> MAX_VELOCITY = 0.35
> MAX_ACCELERATION = 30.0
> MIN_LIMIT = -0.01
> MAX_LIMIT = 8.4
> 
> [JOINT_0]
> TYPE = LINEAR
> HOME = 0.0
> MIN_LIMIT = -0.01
> MAX_LIMIT = 8.4
> MAX_VELOCITY = 0.35
> MAX_ACCELERATION = 30.0
> STEPGEN_MAXACCEL = 37.5
> SCALE = 8000.0
> FERROR = 0.05
> MIN_FERROR = 0.01
> HOME_OFFSET = 0.0
> 
> [AXIS_Y]
> MAX_VELOCITY = 0.35
> MAX_ACCELERATION = 30.0
> MIN_LIMIT = -0.01
> MAX_LIMIT = 3.4
> 
> [JOINT_1]
> TYPE = LINEAR
> HOME = 0.0
> MIN_LIMIT = -0.01
> MAX_LIMIT = 3.4
> MAX_VELOCITY = 0.35
> MAX_ACCELERATION = 30.0
> STEPGEN_MAXACCEL = 37.5
> SCALE = 8000.0
> FERROR = 0.05
> MIN_FERROR = 0.01
> HOME_OFFSET = 0.0
> 
> [AXIS_Z]
> MAX_VELOCITY = 0.15
> MAX_ACCELERATION = 30.0
> MIN_LIMIT = -8.7
> MAX_LIMIT = 0.01
> 
> [JOINT_2]
> TYPE = LINEAR
> HOME = 0.0
> MIN_LIMIT = -8.7
> MAX_LIMIT = 0.01
> MAX_VELOCITY = 0.15
> MAX_ACCELERATION = 30.0
> STEPGEN_MAXACCEL = 37.5
> SCALE = 8000.0
> FERROR = 0.05
> MIN_FERROR = 0.01
> HOME_OFFSET = 0.0

gmoccapy, 2016-10-23:
That will not happen, if you have connected a real machine, wouldn't it?
As if you home the machine, and the INI is set correctly, the soft limits are disabled. And after homing everything should be fine.

The behavior you described is known also in 2.6 or 2.7 branch.

IMHO no ISSUE.

Norbert

jepler, 2016-10-23:
I think it's a bug, just not a gmoccapy bug.  It seems more likely it's a JA bug.

Compare the behavior of 2.7 and master in configs/sim/axis/axis.ini

F1 F2.  Turn jog speed way up.  Hold cursor left to jog in -X.

In 2.7, the jog will continue for at least as long as the total soft limit range in X, but stop at some point.  then, if you press cursor left again, it will continue jogging.

In master, the jog stops right where linuxcnc thinks the soft limit boundaries are.

gmoccapy, 2016-10-23:
Oh, I see that is the problem.
I agree with you, it is a bug!

Kurt-Jacobson, 2017-01-12:
This is certainly a bug, and a very frustrating one for me since I have to take the motors off so I can move the axes home manually as I don't have home switches, yet.

pippin88, 2017-10-20:
I have encountered this behaviour on the current master build.

I have a lathe with travels defined as negative ranges (-245 to 0mm, -95 to 0mm). I cannot jog positive until homing is performed. I.e. I turn on the machine with the axes in the middle of their travel, and I cannot job positive, as LinuxCNC seems to think that that position is 0,0.
I can activate homing and that works fine. However, homing is slow by necessity. I prefer to jog close to my home switches before activating the homing sequence.

grandixximo, 2026-10-04:
Still present on master, root cause found. The joint jog path got this right in 2006 (refresh_jog_limits: unhomed jogs one limit range from the current position), but the axis (teleop) jog path in axis.c clamps to the INI soft limits unconditionally, and update_teleop_with_check() re-imposes them every servo cycle. Only identity kinematics can enter teleop unhomed, which is why the stock GUIs never show it: AXIS and gmoccapy stay in joint mode until homed. It bites axis jog sources wired straight to motion, like a panel on the halui axis pins with NO_FORCE_HOMING=1.

Reproduced on sim/axis (X limits ±10): unhomed teleop jog -X stops at -10.00 instead of jogging one range (-20) from the current position.

Fix in #4631: an axis is homed when every joint mapped to its coordinate letter is homed (gantry letters need all their joints). Homed axis keeps the INI limits, unhomed axis jogs one limit range from its position, same rule as joint jogs. Verified live: unhomed X jog now reaches -20, and with only Z homed, Z jogs to its INI limits while X and Y still jog one range from wherever they stand.

closed 2026-10-11 by grandixximo