LinuxCNC/linuxcnc #200: active G-Code is not updated correctly

gmoccapy, 2016-10-23:
This concern 2.7 and master, not checked on 2.6.

just open any config you like.

- get out of estop
- start machine
- home all axis
- go to mdi 
- enter S1500 M3

active gcode show M3

- push Spindle stop button

spindle stops, but M3 remains active, but should change to M5!

The same happen with cooling button.

Norbert

jepler, 2016-10-23:
Yes, that's how it has worked for quite a long time, possibly forever.  The spindle start/stop buttons work by NML message, which is separate from the interpreter's idea of whether M3/M4/M5 is the active mode.

gmoccapy, 2016-10-23:
Hope you do not mean, that it is OK, because it has been that way for long time.

Norbert

gmoccapy, 2016-10-24:
Jepler, you added a label enhancement, but IMHO it is wrong and should be treated as a bug.

If you give me a advice where I can look at, I will try. I do not know NML nor C++, but I am familiar with python ;-)

Norbert

gmoccapy, 2016-12-07:
Any comments on this from the developer crew?

Norbert

terrylm, 2017-01-10:
This might be related to issue #223 that I just posted?

petterreinholdtsen, 2023-06-21:
I was surprised by this recently, when testing a machine for the first time.  Did not know G-code that well, so I tried to do the first cut using MDI instruction 'G1 S1000 X-120'.  I know now that I forgot M3 to start the spindle, so the spindle did not rotate at all.  Then tried the manual mode to start the spindle and set rotation speed to 1000 rpm, returned to the MDI screen to give the same instruction as before, but alas, the axis started moving while the spindle stopped.  This was quite surprising to me.