LinuxCNC/linuxcnc #203: python interface, settings returns read ahead values and shouldn't

gmoccapy, 2016-10-23:
tested with 2.6, 2.7 and master

if we ask for self.stat.settings we get a tupple as documented, but ...

self.stat.settings[1] returns the read ahead feed value, while
self.stat.settings[2] returns the actual running speed value

shouldn't settings return the actual values for both?

Is there a way to have some attributes like

self.stat.current_feed
self.stat.current_speed

behaving as the self.stat.current_vel?

At the moment there is no way to get the actual feed value from python, at least non I know.

Norbert

fixerslo, 2016-11-09:
The same issue is also with active G-Codes, it shows read-ahead values instead of current values and current program line doesn't return any useful value when there is non-linear program flow, for instance when using subroutines and loops.

cradek, 2016-11-09:
Please help us test the statetags branch, which is mean to fix these problems.

gmoccapy, 2016-11-12:
Hallo Cradek,

What help do you need exactly?
Just getting the branch and do testing?

Norbert

fixerslo, 2017-01-15:
anything new on this issue?