LinuxCNC/linuxcnc #204: docs: LinuxCNC_Documentation.pdf - double chapter

tseufl, 2016-10-27:
In the pdf output of the docs a chapter is duplicated (see picture below).

in:
 -> linuxcnc/docs/src/Master_Documentation.txt <-
line 17-19:

17 = LinuxCNC History
18
19 include::common/emc-history.txt[]

Here is a chapter title (= LinuxCNC History) and the following doc (emc-history.txt) also contains exactly the same title (for html and pdf-output)

This seems to me to be the cause.

Changing the title in emc-history.txt is not a good idea, because then the html and pdf issues can cause problems.
My idea to solve the problem would be to modify the chapter title in linuxcnc / docs / src / Master_Documentation.txt something so that they are no longer exactly the same.

[image]

jethornton, 2018-07-10:
I don't see the double title in master or 2.7

closed 2018-07-10 by jethornton

SebKuzminsky, 2018-07-10:
This was fixed in d0b5fd91a146d4fcf02cbf01fff2c5f8adb3fc32 (in 2.7.8 and later).