LinuxCNC/linuxcnc #207: website: new forum template -> missing link + wrong favicon

tseufl, 2016-10-29:
In the new forum template (kunena 5.0) is nowhere a single link to the mainpage.
The mainpage (including html docs) loses a lot of linkpower for seo and for a user who has landed via a search engine in the forum, it is not easy to get to the mainpage.

Furthermore, the new template uses an incorrect favicon (joomla -> /templates/protostar/favicon.ico).
I had therefore registered in the forum again and already posted it in "Kunena forum questions" ...

gmoccapy, 2016-10-29:
I do miss also a link to mark all topics read.

Norbert

jepler, 2016-10-30:
We (I) administer the forum.  We (I) don't develop the software.

Where possible, we (I) change settings in the administrative interface to improve the experience for users, but I don't do other stuff such as directly edit php or theme files, since these are routinely overwritten by joomla/community builder/kunena updates and 100% "click to keep up to date with php security flaws" is much more important than making cosmetic changes.

So for instance the method joomla documents for changing the favicon will be overwitten the next time joomla updates or changes the default theme: https://docs.joomla.org/changing_the_site_favicon

The "mark all topics read" button disappeared in a Kunena update.  If someone will point me at information about how to re-add the button through the administrator interface, I'll do it.  Otherwise, I'm not going to develop and implement such functionality.

I did add the requested link.

closed 2016-10-30 by jepler

tseufl, 2016-10-30:
I do not know where the joomla core for the kunena extension is and is configured.
In the link (https://docs.joomla.org/changing_the_site_favicon) at the end of the page (last code box):
=> // Try to find a favicon by checking the template and root folder <=
Could be the problem.
In the root linuxcnc.org I can see a favicon.png but I can not find favicon.ico (with correct filetype: .ico). If joomla probably can not find favicon.ico it will probably automatically put into the template the protostar default joomla icon ???

floppydisk525, 2016-10-30:
http://imgur.com/a/x5R3Y

For some reason I couldn't upload a favicon.ico or favicon.zip (zip file w/ favicon in it).  So, I put it on imgur, you should be able to download favicon.ico of chips there.

jepler, 2016-10-30:
@tseufl the point is, the location is one that is overwritten with joomla updates.