LinuxCNC/linuxcnc #211: hm2 dpll: reportedly, setting even-numbered timers doesn't work

jepler, 2016-11-16:
According to pcw (via IRC):
> if you set timer 1 it works
> if you set timer 2 but never set timer 1 timer 2 is never updated 
> same for 3,4
> the default timer values (+100 usec) are not written correctly ether
> timer 1,2 share a 32 bit register as do 3,4

pcw-mesa, 2017-08-11:
so if I use timer 1,  and set timer 1 for -50 usec I get
0xE6670CCC in timerreg12 (I think the 0xE667 is the default 100 usec and 0xCCC is -50)
and everything works as expected
but if i use timer2  and set timer 2 for -50 usec, I get
0xFFFFE667 in timerreg12 so the timer2 value is  not set
If I then set timer 1 to -50 usec i get the expected
0x0CCC0CCC in timerreg12

andypugh, 2017-10-18:
Fixed: commit 66a518e1f8320ba66aacd61ade46782cbfad9fef
The actual error was much worse than described. Even timers could never be set if the odd timers had a positive offset.

closed 2017-10-18 by andypugh