LinuxCNC/linuxcnc #222: Multi-turn arcs incorrectly being converted to straight feeds

robEllenberg, 2017-01-09:
## Here are the steps I follow to reproduce the issue:
Run the following G code snippet in a mill simulation config:
```
G20 G90 G17
G64 P0.005 Q0.001
G0 X 0.0650 Y 0.0000
G0 Z 0.2000
G1 Z 0.0000 F10.0
F 3.42 (Arc Feed, inches/minute)
G3 X 0.0643 Y 0.0095 Z -0.2000 P 15.0000 I -0.0650 J 0.0000
G3 X 0.0643 Y 0.0095 Z -0.2000 P 1.0000 I -0.0643 J -0.0095
F 10.00 (Feed, inches/minute)
G0 Z 0.2000 (Z Clear)
```

## This is what I expected to happen:
15-turn helical bore down to Z-0.2, followed by a full circle finish pass at the bottom of the hole.

## This is what happened instead:
Straight plunge to Z-0.2, then a full-circle finish pass

## It worked properly before this:
Fails on 2.7.x versions (definitely 2.7.8, likely older versions as well).

## Diagnosis
This bug is due to the Naive CAM Detection routine in ARC_FEED in emccanon.cc. NCD checks if the [chord height](https://en.wikipedia.org/wiki/Circular_segment) between an arc segment and a line is below the NCD tolerance. If it is, then it replaces the circular arc with a pair of line segments.  Unfortunately, this check does not properly account for multi-turn arcs. Currently, it ignores the number of turns entirely. What it **should** do is treat all angles more than 180 degrees as 180 deg (i.e. the maximum chord height).

robEllenberg, 2017-01-09:
I have a fix in progress in [this branch](https://github.com/robEllenberg/linuxcnc-mirror/tree/bugfix/issue-222-ncd-multiturn-arcs) in my fork. I haven't tested it much yet, but so far it works for some simple test cases.

andypugh, 2020-06-13:
Is this ready for merge to LinuxCNC?

robEllenberg, 2020-06-13:
I'll need to take another look at this. It worked well under simple conditions, but I ran into some weirdness with tool compensation. It's not a show-stopper as it is, provided users are cautious with NCD.

andypugh, 2020-07-30:
Revert docs note ff1a36631 when this is fixed.