LinuxCNC/linuxcnc #223: Axis: Improper source file refresh when clicking the broom.

terrylm, 2017-01-10:
## Here are the steps I follow to reproduce the issue:

 1. Start axis, home, load some.ngc file
 2. Move the axises (may not be needed.)
 3. Use an externally started editor to add or delete something obvious in the ngc file. Save it.
 4. Click on the broom icon to erase the movement traces in the graphic window.
 5. Notice that the displayed code is unchanged. It does not reflect the edit.
 6. Run the code. Notice that the changes are in effect, though not displayed in the code window.

## This is what I expected to happen:
I would expect that clicking the broom icon would only clean up the graphic display, and NOT reload the source ngc file.

## This is what happened instead:
See above

## It worked properly before this:
Had linuxcnc 2.5.4, did clean install of 2.7.4, installed latest updates, now at 2.7.8.
The issue did not exist in 2.5.4, I am sure I would have noticed.
It started after installing 2.7.4 and continues in 2.7.8.

## Information about my hardware and software:
I used the iso image provided, so it is the standard out-of-the-box install.
 * I am using this kernel version (shown by `uname -a`):
Linux mill 3.4-9-rtai-686-pae #1 SMP PREEMPT Debian 3.4.55-4linuxcnc i686 GNU/Linux
On an old Pentium 4 that is only used for my mill.
 * I am running ...
   * [ X] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
1:2.7.8
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
AXIS
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): 
5i23 Mesa card.

This might be related to issue #200?

jepler, 2017-01-10:
I think that step 4 (clicking the toolbar icon) is actually unnecessary to produce this behavior. At least I've been unable to think of a reason why it would actually influence the outcome...

I don't know what guarantee LinuxCNC has ever offered when the part program is changed between the time it is opened and when it is run (and it's entirely possible that some other internal changes in LinuxCNC have impacted this behavior; different editor programs could potentially cause different behavior in LinuxCNC as well, depending whether they overwrite the original file in place, or remove the original file and replace it with a fresh copy of the newly-written file).

If you prefer that edits are *NOT* seen without a "reload" command, then I think there's a workaround that will work as long as you're using the AXIS UI . It is a trick that uses the "filter" feature of AXIS UI.  (some other UIs may support this feature, but I didn't check any of them)  

Say you prefer to use the ".ngc" extension for your part programs, find the `[FILTER]` section in your inifile and add the line `ngc = cat`.  (`cat` is a unix program that reads a file of input and produces the same file as output)  When you do this, the actual part program that LinuxCNC loads is in effect a copy of the original, and any edits to the original file won't be reflected.  You can list as many extensions as you like, so you could also put `gcode = cat`, `nc = cat`, etc., for any extension you think you are likely to encounter.

terrylm, 2017-01-10:
Hi Jepler

you are correct, no need to click the broom.
I will use cat as you suggest. Thanks

It does still seem to me to be a bug though.