LinuxCNC/linuxcnc #224: Soft limits while unhomed

lumberjackjm, 2017-01-18:
## Here are the steps I follow to reproduce the issue:

 1. switch on power to my gantry router (no home or limit switches installed)
 2. launch axis 2.8.0
 3. (tool is located in some arbitrary xy location above the 48" x 96" bed)
 3. jog machine to home position

## This is what I expected to happen:
In previous version, I am allowed to jog to home location without the constraints of the soft limits programmed into the .ini file. Soft limits only take effect after axis homing.

## This is what happened instead:
I am prevented from jogging past my lower limit (-1" in my case)

## It worked properly before this:
upgrading to 2.8.0

## Information about my hardware and software:

 * I am using Debian 7.11
 * I am using this kernel version (3.4-9-rtai-686-pae #1 SMP PREEMPT Debian 3.4.55-4linuxcnc i686 GNU/Linux
 * I am running ...
   * A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   
 * I am using this LinuxCNC version (2.8.0):
 * I am using this user interface (AXIS):
 * I am using this interface hardware vendor and chipset (parallel port):

pippin88, 2017-10-20:
This appears to be the same issue as #182 (https://github.com/LinuxCNC/linuxcnc/issues/182)

grandixximo, 2026-10-04:
This is the same root cause as #182: the axis (teleop) jog path clamps to the INI soft limits even when unhomed, treating the unknown position as 0. Joint jogs use one limit range from the current position instead. Still present on master.

Fix in #4631, with per-axis homed checks: a homed axis keeps its INI limits, an unhomed axis jogs one limit range from its position. Closing this as a duplicate of #182; the PR will close both when merged.

grandixximo, 2026-11-01:
Fixed by #4631 (merged 2026-10-11): the teleop jog path now checks homed state per axis, so an unhomed axis jogs one limit range from its current position instead of being clamped to the INI limits as if at 0. Closing as duplicate of #182.

closed 2026-11-01 by grandixximo