LinuxCNC/linuxcnc #225: Transfer simple-gcode-generators to linuxcnc org

njh, 2017-01-24:
Hello,

Sorry, this doesn't seem quite the right place but wanted to get someone in the linuxcnc GitHub organisation.

As discussed on the emc-developers, I have make a Git repo out of the simple Python G-Code generators:
https://sourceforge.net/p/emc/mailman/emc-developers/thread/4b3fd6f50cc412093608bd7fabe1086b%40mail.aelius.com/

The repo is currently here:
https://github.com/njh/simple-gcode-generators

Would someone be able to help me move this to the linuxcnc org:
https://github.com/LinuxCNC

I think I either need to be a member of the org, or transfer it to an individual who is a memeber of the org first.

Thanks,

nick.

jepler, 2017-02-04:
It looks like we have to do this with several steps:

 1. @njh initiates a transfer of simple-gcode-generators to @jepler (or any other person with admin membership in the LinuxCNC organization, but I'll be happy to take care of it)
 1. @jepler transfers it to the LinuxCNC organization on github
 1. @jepler fixes up the other details of repo permissions so that @njh has e.g., permissions to manage the issue tracker  (and any other github collaborators as appropriate)

Doing this in multiple steps means avoids making @njh an administrator *for the LinuxCNC organization*.  To do tasks like manage the issue tracker for simple-gcode-generators, we can use *teams*, giving team members *write* access at the organization level.  I am not fully up to speed on how all of this works, but I've been reading at
 * https://help.github.com/articles/transferring-a-repository-owned-by-your-personal-account/
 * https://help.github.com/articles/repository-permission-levels-for-an-organization/
and I think this will work as desired.

njh, 2017-02-04:
Thank you very much @jepler, was starting to worry that it had all been wasted effort 😨 

I have initiated the transfer to you.

jepler, 2017-02-04:
hm I'm not seeing the transfer of simple-gcode-generators from @njh to @jepler yet.  did you complete all 5 steps at https://help.github.com/articles/transferring-a-repository-owned-by-your-personal-account/ or did you run into trouble?

jepler, 2017-02-04:
aha found it (the notification goes to email only, apparently)

jepler, 2017-02-04:
https://github.com/LinuxCNC/simple-gcode-generators

I believe that @njh should have permission to push to the repo, manage issues, and so forth.  For some activities (like adding new collaborators to the simple-gcode-generators team) it'll be necessary to contact one of the admins of the LinuxCNC organization on github, including @jepler @SebKuzminsky and @cradek; opening issues on LinuxCNC/linuxcnc is a fine way to do this, probably better than mailing list, IRC, or forum.

closed 2017-02-04 by jepler

njh, 2017-02-05:
Just checked and I indeed have permission to push to the repo.

Looking good, thanks for sorting this out @jepler 😃