LinuxCNC/linuxcnc #228: Build version 2.7.8 on i586 failed

midyukov-anton, 2017-02-03:
Build version 2.7.6-2.7.8 on i586 failed.
```
Compiling realtime objects/hal/components/clarke2.c
Linking ../rtlib/clarke2.so
`__x86.get_pc_thunk.dx' referenced in section `.text' of objects/clarke2.tmp: defined in discarded section `.text.__x86.get_pc_thunk.dx[__x86.get_pc_thunk.dx]' of objects/clarke2.tmp
collect2: error: ld returned 1 exit status
make: *** [../rtlib/clarke2.so] Error 1
```

[build_2.7.8-i586.txt](https://github.com/LinuxCNC/linuxcnc/files/749547/build_2.7.8-i586.txt)

midyukov-anton, 2017-02-03:
In version 2.7.5 also reproduced. Distribution ALT Linux. Prompt, please, in what could be the problem?

SebKuzminsky, 2017-02-03:
Neat, RPMs, that's new.

We regularly build 2.7 for uspace on 32-bit x86, and I've never seen this problem.  My guess is that it's caused by some difference in our regular build environment and this RPM build environment you're working with.  

What version of GCC are you using?

What commands does your build system run to do all this?  At least the configure steps look quite different than our regular build (described here: http://linuxcnc.org/docs/devel/html/code/building-linuxcnc.html).  There are also some errors during configure, though I don't know if they're related to the fatal error later.

midyukov-anton, 2017-02-03:
>What version of GCC are you using? 

Used gcc6. But the problem first came in August, when trying to build version 2.7.6. Used when gcc5. I try to collect a gcc5 version 2.7.5.

> What commands does your build system run to do all this?

link to the spec: http://git.altlinux.org/tasks/177559/gears/40/git?p=git;a=blob;f=linuxcnc.spec;h=5f5a0aa16a4e8ec16764782ab1761281ae0c74a3;hb=ab589e25c49e2d1abcaa3ca9d6ffdb5c87844fbb

The main difference is that in the assembly, I instead use autogen.sh autoreconf -fisv

SebKuzminsky, 2017-02-03:
I build LinuxCNC 2.7 with GCC 6 (on Debian Stretch) all the time without any trouble.

SebKuzminsky, 2017-02-03:
I get a 404 on the git.altlinux.org link in your comment.

jepler, 2017-02-03:
Whatever exactly is going on probably needs to be fixed near here: https://github.com/LinuxCNC/linuxcnc/blob/master/src/Makefile#L995

We play weird games with symbols in the linker to get semantics similar to kernel symbol exporting in userspace. There used to be a special case for get_px_think symbols but we thought the current iteration was right without losing all possible thinks for all possible architectures. Apparently not.

SebKuzminsky, 2017-02-03:
I can reproduce this problem on Debian Stretch, 32-bit x86 installation, with GCC 6.3.0-5.  The master branch builds fine but 2.7 fails with a similar error as what @midyukov-anton.

SebKuzminsky, 2017-02-03:
The diff from 2.7 (which fails) to master (which does not fail) in the area @jepler pointed to is this:

```
@@ -960,14 +988,13 @@ RTOBJS := $(sort $(foreach mod,$(obj-m),$(call TORTOBJS,$(mod))))
 
 RTDEPS := $(sort $(RTOBJS:.o=.d))
 
-IS_POWERPC = test `uname -m` = ppc -o `uname -m` = ppc64
 modules: $(patsubst %.o,../rtlib/%.so,$(obj-m))
 ../rtlib/%.so:
        $(ECHO) Linking $@
        $(Q)ld -d -r -o objects/$*.tmp $^
-       @if ! $(IS_POWERPC); then objcopy -j .rtapi_export -O binary objects/$*.tmp objects/$*.exp; fi
-       @if ! $(IS_POWERPC); then objcopy -G __x86.get_pc_thunk.bx -G __i686.get_pc_thunk.bx `xargs -r0n1 echo -G < objects/$*.exp | grep -ve '^-G $$' | sort -u` objects/$*.tmp; fi
-       $(Q)$(CC) -shared -Bsymbolic $(LDFLAGS) -o $@ objects/$*.tmp -lm
+       $(Q)objcopy -j .rtapi_export -O binary objects/$*.tmp objects/$*.sym
+       $(Q)(echo '{ global : '; tr -s '\0' < objects/$*.sym | xargs -r0 printf '%s;\n' | grep .; echo 'local : * ; };') > objects/$*.ver
+       $(Q)$(CC) -shared -Bsymbolic $(LDFLAGS) -Wl,--version-script,objects/$*.ver -o $@ $^ -lm
 
 RTFLAGS += -fno-strict-aliasing -fwrapv
 
```

The second objcopy is missing, which looks suspicious, but I don't know this part of the code well enough to hazard a guess.

jepler, 2017-02-03:
@SebKuzminsky ah so it's fixed in master branch? That's great. Maybe we should just back port the change

SebKuzminsky, 2017-02-04:
I cherrypicked @jepler's fix (f87c0244) from master to a new branch called 2.7-rtlibs-fix.  That fix lets the compile succeed on my Stretch i386 machine.

It was not a straight-forward cherrypick, and I'd appreciate a sanity-check from @jepler.  In the meanwhile, the buildbot is testing it on our current platforms here: http://buildbot.linuxcnc.org/buildbot/builders/0000.checkin/builds/4739

@midyukov-anton, please give that branch a try.

midyukov-anton, 2017-02-04:
Yesterday I was able to gather here with the patch, just in my time zone was evening, and I went to sleep.
```
diff --git a/src/Makefile b/src/Makefile
index 2e903e4..9da3e10 100644
--- a/src/Makefile
+++ b/src/Makefile
@@ -966,7 +966,7 @@ modules: $(patsubst %.o,../rtlib/%.so,$(obj-m))
 	$(ECHO) Linking $@
 	$(Q)ld -d -r -o objects/$*.tmp $^
 	@if ! $(IS_POWERPC); then objcopy -j .rtapi_export -O binary objects/$*.tmp objects/$*.exp; fi
-	@if ! $(IS_POWERPC); then objcopy -G __x86.get_pc_thunk.bx -G __i686.get_pc_thunk.bx `xargs -r0n1 echo -G < objects/$*.exp | grep -ve '^-G $$' | sort -u` objects/$*.tmp; fi
+	@if ! $(IS_POWERPC); then objcopy -G __x86.get_pc_thunk.bx -G __i686.get_pc_thunk.bx -G __x86.get_pc_thunk.dx `xargs -r0n1 echo -G < objects/$*.exp | grep -ve '^-G $$' | sort -u` objects/$*.tmp; fi
 	$(Q)$(CC) -shared -Bsymbolic $(LDFLAGS) -o $@ objects/$*.tmp -lm
 
 RTFLAGS += -fno-strict-aliasing -fwrapv
diff --git a/src/Makefile.modinc.in b/src/Makefile.modinc.in
index 5ee051c..c0ab54d 100644
--- a/src/Makefile.modinc.in
+++ b/src/Makefile.modinc.in
@@ -120,7 +120,7 @@ IS_POWERPC = test `uname -m` = ppc -o `uname -m` = ppc64
 	$(ECHO) Linking $@
 	$(Q)ld -d -r -o $*.tmp $^
 	$(Q)if ! $(IS_POWERPC); then objcopy -j .rtapi_export -O binary $*.tmp $*.exp; fi
-	$(Q)if ! $(IS_POWERPC); then objcopy -G __x86.get_pc_thunk.bx -G __i686.get_pc_thunk.bx `xargs -r0n1 echo -G < $*.exp | grep -ve '^-G $$' | sort -u` $*.tmp; fi
+	$(Q)if ! $(IS_POWERPC); then objcopy -G __x86.get_pc_thunk.bx -G __i686.get_pc_thunk.bx -G __x86.get_pc_thunk.dx `xargs -r0n1 echo -G < $*.exp | grep -ve '^-G $$' | sort -u` $*.tmp; fi
 	$(Q)$(CC) -shared -Bsymbolic $(LDFLAGS) -o $@ $*.tmp -lm
 endif
```
 
Today, I check your patch
Yesterday found here is a discussion on the use of objcopy: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=59842#c4

midyukov-anton, 2017-02-04:
2.7-rtlibs-fix building successful. Thanks!

jepler, 2017-02-04:
@SebKuzminsky I'd just take the whole text of that rule from master branch into 2.7 rather than trying to cherry-pick -- it looks like it went through a couple of revisions due to the docker bug that an intermediate version was triggering.  In particular, I'm surprised to see any checks of IS_POWERPC left in that Makefile rule.

commit b1157204 (2017-02-06): fix link error on i686 with gcc, or maybe objcopy 2.27

This commit fixes #228, reported by Anton Midyukov.  Adapted for 2.7
from Jeff Epler's commit f87c0244.

On Debian Stretch on i686, building 2.7 results in rtlibs link-time
failures like this:

    Compiling realtime objects/hal/components/clarke2.c
    Linking ../rtlib/clarke2.so
    `__x86.get_pc_thunk.dx' referenced in section `.text' of objects/clarke2.tmp: defined in discarded section `.text.__x86.get_pc_thunk.dx[__x86.get_pc_thunk.dx]' of objects/clarke2.tmp
    collect2: error: ld returned 1 exit status
    make: *** [../rtlib/clar [... cut]
closed 2017-02-06 by cradek